markwal/OctoPrint-PolarCloud #1: Install fails during dependency checks...

fiveangle, 2017-06-25:
Using _OctoPrint 1.3.4 (master branch)_, install fails:

```
Installing plugin "PolarCloud (Beta)" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
/home/pi/oprint/bin/python -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg (from OctoPrint-PolarCloud==0.1.0Beta)
Collecting SocketIO-client (from OctoPrint-PolarCloud==0.1.0Beta)
Using cached socketIO-client-0.7.2.tar.gz
Requirement already satisfied: pyopenssl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==0.1.0Beta)
Exception:
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
status = self.run(options, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 324, in run
requirement_set.prepare_files(finder)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
ignore_dependencies=self.ignore_dependencies))
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 666, in _prepare_file
check_dist_requires_python(dist)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/packaging.py", line 48, in check_dist_requires_python
feed_parser.feed(metadata)
File "/usr/lib/python2.7/email/feedparser.py", line 177, in feed
self._input.push(data)
File "/usr/lib/python2.7/email/feedparser.py", line 99, in push
parts = data.splitlines(True)
AttributeError: 'NoneType' object has no attribute 'splitlines'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

<details><summary>plugin manager console output log</summary>

```
2017-06-25 03:15:56,948   /home/pi/oprint/bin/python -m pip install https://github.com/Booli/OctoPrint-ActionTriggerPlugin/archive/master.zip
2017-06-25 03:16:00,983 > Collecting https://github.com/Booli/OctoPrint-ActionTriggerPlugin/archive/master.zip
2017-06-25 03:16:02,201 > Downloading https://github.com/Booli/OctoPrint-ActionTriggerPlugin/archive/master.zip
2017-06-25 03:16:08,001 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg (from OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,002 > Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,003 > Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,003 > Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,004 > Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,004 > Requirement already satisfied: sockjs-tornado<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,005 > Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,005 > Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,006 > Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,006 > Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,007 > Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,008 > Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.7-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,008 > Requirement already satisfied: pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,009 > Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.17-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,009 > Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,010 > Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,010 > Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.0-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,011 > Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,011 > Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,012 > Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,013 > Requirement already satisfied: requests<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,013 > Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,014 > Requirement already satisfied: psutil<3.3,>=3.2.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-3.2.1-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,014 > Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,015 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,015 > Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,016 > Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,016 > Requirement already satisfied: future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,017 > Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1)
2017-06-25 03:16:08,017 > Requirement already satisfied: websocket-client<0.41,>=0.40 in /home/pi/oprint/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-ActionTriggerPlugin==0.1 [... cut ...]
2017-06-25 04:12:56,191 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/packaging.py", line 48, in check_dist_requires_python
2017-06-25 04:12:56,191 ! feed_parser.feed(metadata)
2017-06-25 04:12:56,192 ! File "/usr/lib/python2.7/email/feedparser.py", line 177, in feed
2017-06-25 04:12:56,192 ! self._input.push(data)
2017-06-25 04:12:56,192 ! File "/usr/lib/python2.7/email/feedparser.py", line 99, in push
2017-06-25 04:12:56,193 ! parts = data.splitlines(True)
2017-06-25 04:12:56,193 ! AttributeError: 'NoneType' object has no attribute 'splitlines'```
</details>

KayIO, 2017-06-25:
Yes ... Same with me.

dnewman-polar3d, 2017-06-25:
FWIW,

With a fresh Octopi SD (2016-03-18-octopi-jessie-lite-0.13.0.img) on a RPi 3, the install worked for me.  However, I first

1. Allowed Octoprint to update to 1.3.4,
2. Installed the GPX plugin (not required in general),
3. Manually installed GStreamer (which required two passes as I had not previously done an "apt-get update" which turned out to be needed), and then
4. Installed the Polar Cloud plugin.  That, for me at least on Raspbian, went smoothly.

The manual install of GStreamer would not seem to be relevant to the logged error cited above though.

markwal, 2017-06-25:
Apparently this is an interaction between pyOpenSSL and pip:

https://github.com/guysoft/OctoPi/issues/276

markwal, 2017-06-25:
OK.  I've verified again that these steps work for installation purposes:

* Make a new fresh image of octopi 0.13
* Update to OctoPrint 1.3.4
* Install the plugin

So, I suspect that people that hit this install issue are those that have the newest pip (9.0.1) because they've updated it after seeing the warning at the end of every plugin install.  I'm going to try that next.

dnewman-polar3d, 2017-06-25:
I didn't have the newest pip.....

Again, I started with
1. A fresh Octopi SD (2016-03-18-octopi-jessie-lite-0.13.0.img) on a RPi 3
2. Allowed Octoprint to update to 1.3.4,
3. Installed the GPX plugin (not required in general),
4. Manually installed GStreamer (which required two passes as I had not previously done an "apt-get update" which turned out to be needed), and then
5. Installed the Polar Cloud plugin. That, for me at least on Raspbian, went smoothly.

I didn't update pip until much later; when I went to manually remove and re-install pyOpenSSL by hand.

markwal, 2017-06-25:
Oh boy.  With the newest pip, it doesn't look like the workaround works.  I can't get pyOpenSSL to install at all.  :-(

dnewman-polar3d, 2017-06-26:
The updated directions do not work for me.  I note also that I did not initially have pyOpenSSL to remove with the command,

    rm -rf ~/oprint/lib/python2.7/site-packages/pyOpenSSL*

Perhaps that should have been

    rm -rf ~/oprint/lib/python2.7/site-packages/OpenSSL*

markwal, 2017-06-26:
Maybe both?

Steveveepee, 2017-06-26:
I had errors still after following the Readme.md 
I manually removed the OpenSSL folder with `rm -rf ~/oprint/lib/python2.7/site-packages/OpenSSL` and then it worked.

markwal, 2017-06-27:
I've reworked this so now the plugin falls back to using "ssh-keygen" if the pyOpenSSL module is too old to have the method I need, so the workaround steps are no longer necessary on systems that have ssh-keygen available.  However, Windows users will still need to manually upgrade pyOpenSSL to at least 16.1 since they're less likely to have ssh-keygen on their path (unless they've installed Git for Windows).

closed 2017-06-27 by markwal