markwal/OctoPrint-PolarCloud #2: Failed to register OctoPrint with the Polar Cloud.

howsTricks, 2017-06-25:
When I try to register I get the following error message:

Failed to register OctoPrint with the Polar Cloud.

I've enabled Verbose logging but don't get any additional output, is it stored to a specific file?

dnewman-polar3d, 2017-06-25:
FWIW Mark, no connection attempt was logged in the cloud.  Everything related to registering is logged at INFO level logging and thus should be logged should it be attempted, failure or success.  So this suggests things did not manage to connect to https://printer2.polar3d.com.

howsTricks, 2017-06-25:
Looks like it has no issue, this is from the pi:

pi@octopi-1:~ $ traceroute printer2.polar3d.com
traceroute to printer2.polar3d.com (54.201.106.190), 30 hops max, 60 byte packet                                                                 s
 1  REMOVED
 2  REMOVED
 3  A357-gw-c.bahnhof.net (46.59.117.22)  4.824 ms  4.743 ms  4.645 ms
 4  A357-gw.bahnhof.net (46.59.117.21)  4.527 ms  4.870 ms  4.791 ms
 5  ste-cgn-ar2.sto-cr1.bahnhof.net (46.59.112.24)  5.536 ms  5.457 ms  5.371 ms
 6  7-1-1.ear2.Stockholm2.Level3.net (213.242.69.193)  4.027 ms  4.342 ms  4.217                                                                  ms
 7  * * *
 8  4.16.146.90 (4.16.146.90)  184.025 ms  183.942 ms  183.857 ms
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  *^C
pi@octopi-1:~ $ wget https://printer2.polar3d.com
--2017-06-25 15:24:38--  https://printer2.polar3d.com/
Resolving printer2.polar3d.com (printer2.polar3d.com)... 54.201.106.190, 54.187.138.189
Connecting to printer2.polar3d.com (printer2.polar3d.com)|54.201.106.190|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-06-25 15:24:39 ERROR 404: Not Found.

pi@octopi-1:~ $ \

dnewman-polar3d, 2017-06-25:
wget won't do much for you as it's not truly an HTTP server.  But you can point your browser at [https://printer2.polar3d.com/info ](https://printer2.polar3d.com/info) and you should get the current date and time in UTC, as served out by the server itself running there.  (It's a single process running under Node.js and primarily using WebSockets.  But it does handle that one HTTPS request for purposes of health monitoring.)

howsTricks, 2017-06-25:
This is what I get
pi@octopi-1:~ $ wget https://printer2.polar3d.com/info
--2017-06-25 15:36:55--  https://printer2.polar3d.com/info
Resolving printer2.polar3d.com (printer2.polar3d.com)... 54.187.138.189, 54.201.                                     106.190
Connecting to printer2.polar3d.com (printer2.polar3d.com)|54.187.138.189|:443...                                      connected.
HTTP request sent, awaiting response... 200 OK
Length: 19 [text/html]
Saving to: ‘info’

info                100%[=====================>]      19  --.-KB/s   in 0s

2017-06-25 15:36:56 (1.67 MB/s) - ‘info’ saved [19/19]

Crowlord, 2017-06-25:
Same issue.

dnewman-polar3d, 2017-06-25:
2017-06-25 15:44:22,278 - octoprint.plugins.polarcloud - INFO - on_api_command {u'command': u'register', u'email': u'dnewman@polar3d.com', u'pin': u'4977183'}
2017-06-25 15:44:22,280 - octoprint.plugins.polarcloud - INFO - emit register
2017-06-25 15:44:22,284 - octoprint - ERROR - Exception on /api/plugin/polarcloud [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/util/flask.py", line 1106, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 853, in on_api_command
    if self._register(data['email'], data['pin']):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 599, in _register
    "publicKey": crypto.dump_publickey(crypto.FILETYPE_PEM, self.key),
AttributeError: 'module' object has no attribute 'dump_publickey'
2017-06-25 15:44:22,310 - tornado.access - ERROR - 500 POST /api/plugin/polarcloud (127.0.0.1) 49.88ms
2017-06-25 15:45:37,817 - octoprint.plugins.polarcloud - INFO - on_api_command {u'command': u'register', u'email': u'dnewman@polar3d.com', u'pin': u'4977183'}
2017-06-25 15:45:37,819 - octoprint.plugins.polarcloud - INFO - emit register
2017-06-25 15:45:37,821 - octoprint - ERROR - Exception on /api/plugin/polarcloud [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/util/flask.py", line 1106, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 853, in on_api_command
    if self._register(data['email'], data['pin']):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 599, in _register
    "publicKey": crypto.dump_publickey(crypto.FILETYPE_PEM, self.key),
AttributeError: 'module' object has no attribute 'dump_publickey'
2017-06-25 15:45:37,827 - tornado.access - ERROR - 500 POST /api/plugin/polarcloud (127.0.0.1) 25.69ms
2017-06-25 15:46:04,650 - octoprint.plugins.polarcloud - INFO - on_api_command {u'command': u'register', u'email': u'dnewman@polar3d.com', u'pin': u'4977183'}
2017-06-25 15:46:04,652 - octoprint.plugins.polarcloud - INFO - emit register
2017-06-25 15:46:04,654 - octoprint - ERROR - Exception on /api/plugin/polarcloud [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
[... cut ...]
2017-06-25 15:46:04,659 - tornado.access - ERROR - 500 POST /api/plugin/polarcloud (127.0.0.1) 24.74ms
2017-06-25 15:46:30,464 - octoprint.plugins.GPX - INFO - Open /home/pi/.octoprint/data/GPX/gpx.ini
2017-06-25 15:46:30,465 - octoprint.plugins.GPX - INFO - Write /home/pi/.octoprint/data/GPX/gpx.ini
2017-06-25 15:46:30,903 - octoprint.plugins.polarcloud - DEBUG - _update_local_settings
2017-06-25 15:46:31,596 - octoprint.plugins.polarcloud - DEBUG - hello
2017-06-25 15:46:31,596 - octoprint.plugins.polarcloud - DEBUG - skip emit hello, serial: None

dnewman-polar3d, 2017-06-25:
Mind you, the OpenSSL module should have crypto.dump_publickey().  I'll leave this for Mark to ponder.  But at any rate, this explains why the cloud isn't seeing a registration command come in: this error is occuring as part of the plugin's attempt to issue that command.  It fails; no command issued.

dnewman-polar3d, 2017-06-25:
Reminder to self: that dump contains one's PIN in cleartext.  Sigh.  Well, it's easy to change at polar3d.com via the account settings page...

mwafler, 2017-06-25:
I'm having the identical issue as well.

howsTricks, 2017-06-25:
Looks like crypto.dump_publickey() was added in the 0.16 release of pyOpenSSL. My (default) installed version is 0.11

Trying to update now to see if it fixes it.

howsTricks, 2017-06-25:
Well, not sure what I did but now it's working. Here's what I did, but not sure what fixed it...

1. Tried updating pyOpenSSL
2. Noticed it had the following error on install: 
IOError: [Errno 2] No such file or directory: '/home/pi/oprint/lib/python2.7/site-packages/OpenSSL/__init__.py'
3. Noticed alot of symlinked files left around referring to the older pyOpenSSL
4. Uninstall pyOpenSSL - Twice, to attempt to get rid of the symlinked pyOpenSSL
5. Manually deleted the left over symlinks
6. Confirmed no pyOpenSSL installed using ~/oprint/bin/pip list
7. Tried registering, seemingly worked? - Which is weird without having the pyOpenSSL package...
8, Installed pyOpenSSL again just for good measure, seems to still be working but not sure if that's due to it already being registered.

Conclusion: There is something very weird with the pyOpenSSL package, note the symlinks in this directory (before I deleted it)

pi@octopi-1:~/oprint/local/lib/python2.7/site-packages/OpenSSL $ ls -al
total 156
drwxr-xr-x  3 pi pi  4096 Apr 18  2016 .
drwxr-xr-x 83 pi pi  4096 Jun 25 16:58 ..
-rw-r--r--  1 pi pi 69376 Feb 27  2014 crypto.arm-linux-gnueabihf.so
lrwxrwxrwx  1 pi pi    46 Feb 27  2014 __init__.py -> ../../../../share/pyshared/OpenSSL/__init__.py
-rw-r--r--  1 pi pi   949 Apr 18  2016 __init__.pyc
-rw-r--r--  1 pi pi 11236 Feb 27  2014 rand.arm-linux-gnueabihf.so
-rw-r--r--  1 pi pi 50540 Feb 27  2014 SSL.arm-linux-gnueabihf.so
drwxr-xr-x  2 pi pi  4096 Apr 18  2016 test
lrwxrwxrwx  1 pi pi    43 Feb 27  2014 tsafe.py -> ../../../../share/pyshared/OpenSSL/tsafe.py
-rw-r--r--  1 pi pi  1649 Apr 18  2016 tsafe.pyc
lrwxrwxrwx  1 pi pi    45 Feb 27  2014 version.py -> ../../../../share/pyshared/OpenSSL/version.py
-rw-r--r--  1 pi pi   253 Apr 18  2016 version.pyc

dnewman-polar3d, 2017-06-25:
Yes, I too noticed that it shipped with just the pyc files and the original sources were invalid symlinks.

markwal, 2017-06-25:
Hmmm...  There's probably a pyOpenSSL installed in the system wide library and apparently a left over one in the virtual environment.  I'd better put a minimum version requirement in the plugin and see if that'll convince pip to upgrade the virtual environment properly.  I didn't see this problem on my test octopi's though, but perhaps I'd already fixed by pyOpenSSL modules for some reason.

debsahu, 2017-06-25:
same issue on my end too.

howsTricks, 2017-06-25:
Comparing between the 0.13.0 img and the 0.14.0 img RC (https://github.com/guysoft/OctoPi/issues/367) looks like the pyOpenSSL files are now no longer included in the \home\pi\oprint\lib\python2.7\site-packages\ folder which is a good thing I think. @guysoft

Crowlord, 2017-06-25:
Have run the extra steps outlined on main page and some updating did occur.

It did claim to have pyOpenSSL already.

My original install was octopi 0.13.0 and octoprint version is 1.3.4

Problem still exists. 

Final install messaged below

=====================
pip install pyOpenSSL
Requirement already satisfied (use --upgrade to upgrade): pyOpenSSL in ./oprint/lib/python2.7/site-packages
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

markwal, 2017-06-25:
I've updated the instructions.  It appears some nuking is necessary.

Crowlord, 2017-06-26:
Im sorry to report that even with a follow of the new nuke procedures (which progressed as expected) and a full reboot the problem still exists :(

dnewman-polar3d, 2017-06-26:
Nuked pyOpenSSL* and OpenSSL*?

Note that there may be a bug in the plugin whereby you need to click "register" twice.  I.e., it can fail the first time but then work the second. 

Also, from the logs, I've seen some cases where people may be looking up their account PIN, changing it, not clicking the SAVE button in the Polar Cloud, and then trying to use that changed PIN.  Which then fails.

Crowlord, 2017-06-26:
Ahhh pretty sure I didn't nuke OpenSSL. I must have missed that. 

I will report back later, just started a print.

debsahu, 2017-06-26:
I second @Crowlord . I nuked pyOpenSSL just as the new instructions say, same issue. The PIN is saved and entered exactly as the account informations reads.

dnewman-polar3d, 2017-06-26:
There is more than one way to skin the cat here.  Instead of using crypto.dump_publickey(), fork out a little shell script and slurp back in the results.   Definitely not as elegant, but....

    cp ~/.octoprint/data/polarcloud/p3d_key ./foo.key
    chmod 0600 ./foo.key
    ssh-keygen -e -m PEM -f ./foo.key > ./pub.key

The resulting file, `./pub.key`, is indeed the correct public key.  At least when I compare my one data point with what was stored in the Polar Cloud.  (Visible under the PRINTER SETTINGS tab of the printer management page; you can access that page from the MANAGE icon on the printer detail/dashboard page.)

By going this more cumbersome route, the need for a newer pyOpenSSL may be obviated.

Note the `chmod 0600` is to keep ssh-keygen from complaining.   In general, pyOpenSSL should have saved that private key with something other than group and world read persmissions.  That seems like a bad idea in pyOpenSSL, but it's not like I've read their docs or anything so it's not really my place to criticize.

Crowlord, 2017-06-26:
Success! This is following the front page instructions and nuking OpenSSL and pyOpenSSL.

dnewman-polar3d, 2017-06-26:
Great.  I sent you an e-mail (x2) as well.   For some reason, I used my personal e-mail and not my @polar3d.com one.   Should the need arise, feel free to contact me using either.  The polar3d.com one: just glue 'dnewman' to the front.

markwal, 2017-06-26:
I was trying to avoid ssh-key because it's not available on Windows.  But I could check for dump_publickey and if it isn't there, try ssh-keygen.

dnewman-polar3d, 2017-06-26:
... if dump_publickey() present, use it.
    ... if not present and platform is OS X or Unix/Linux variant, attempt shell script using ssh-keygen
    ... tell user to take emergency nuking procedures

Note that if the user has installed the git tools on Windows, they have ssh-keygen available via git-bash.  May also be buried elsewhere in the git tools.  It's there for the obvious reason: you often need it with github.   I suppose a sufficient sophisticated shell script (.bat script?) could even try to work some of this out on Windows.

markwal, 2017-06-27:
As of e512b77f5f4, I believe registration should be more reliable.  Several fixes went in:

As discussed, ssh-keygen is now used as a fallback
Also, the plugin disconnects and reconnects after a successful register

closed 2017-06-27 by markwal