markwal/OctoPrint-PolarCloud #4: No video stream

howsTricks, 2017-06-25:
Video stream area (I'm assuming that's what it is) on the Polar Cloud interface shows a grey box with the broken image icon in it in the upper left. I also see a red camera icon on the upper right. I've installed all of the required packages through apt-get and restarted my pi but not getting any video. Confirmed I can see video on my local octoprint interface.

markwal, 2017-06-25:
The plugin has to modify your webcam stream url so that it can show up in another webpage.  The default for octopi is (in OctoPrint->Settings->Webcam/Timelapse->Stream URL

/webcam/?action=stream

The plugin attempts to translate this to:

http://<ipaddress>/webcam/?action=stream

But can have trouble figuring out what the ipaddress is especially if you've got more than one network interface (wifi and ethernet, for example)

You can help it out by giving a full url in that box that works on your network.  For example,

http://<ipaddress>/webcam/?action=stream

Outside of your network, PolarCloud will just show snapshots that are updated every once in a while.

dnewman-polar3d, 2017-06-25:
My guess is that the plugin is tell the cloud to use "/webcam/?action=stream" as the URL for the camera.  That is then fed to the SPA running in your webbrowser but it likely won't work -- it most certainly needs to be an absolute URL.  (However, I could make the backend detect that it's not absolute and glue your local IP to the front....)

howsTricks, 2017-06-25:
Tried changing the Stream URL to `http:///webcam/?action=stream` but get the following in OctoPrint (note the broken image icon)
[image]

howsTricks, 2017-06-25:
I get the same thing as before in the PolarCloud interface:

[image]

dnewman-polar3d, 2017-06-25:
Mark, let me know if there's anything I might do in the web server backend.  While I can pre-pend "http://<local-ip>" to relative URLs, that presumes the service is on port 80.  However, it's no big deal to do that....  I could deploy that this evening in production (and sooner on test beds).

dnewman-polar3d, 2017-06-25:
Indeed, I definitely have to refresh the production systems tonight anyway.  Just discovered a typo in an html page which is causing filament diameter to not appear as one of the slicer config settings.... Ooops!  (Fortunately, it's the sort of frontend issue I'm qualified to diagnose and fix!  It came about as a result of some changes to support XYZ printers.)

markwal, 2017-06-25:
@howsTricks That url has to actually have your octopi's ip address in it:

http://192.168.1.47/webcam/?action=stream

And it has to work with the Test button also.

Don't use 192.168.1.47  use the number (or name) from your address bar.

howsTricks, 2017-06-25:
Ah, that makes more sense... Updated the URL to my OctoPrint IP, test shows the webcam (good) but the PolarCloud interface still shows the broken image icon.

markwal, 2017-06-25:
You *might* have to restart octoprint after updating that URL for my plugin to register the new url with Polar Cloud.  And then you'll need to refresh the polar cloud page (give it a minute after restarting OctoPrint).

dnewman-polar3d, 2017-06-25:
Agreed: you need to restart octoprint.  Either that or kill your network connection and restart it.  When reconnecting to the cloud, the revised URL is sent.  And then things are happy.

[image]

But there's room for improvement... Don't want to have to change that setting in Octoprint each time your printer slurps a new and different IP via DHCP.

howsTricks, 2017-06-25:
There we go! Thanks!

[image]

markwal, 2017-06-27:
I believe with b2e1caf1, I'm now normalizing the url for the stream (might still guess the IP address wrong though...)

closed 2017-06-27 by markwal

3dfernando, 2020-10-21:
I still can't get any stream, even though I use the full local URL: http://10.0.0.226/webcam/?action=stream