markwal/OctoPrint-PolarCloud #6: Bug: Not sending Status 13 when USB not connected + GPX-plugin

dnewman-polar3d, 2017-06-25:
Looks like a status of 0 is being sent when USB is not connected and using the GPX plugin.  This is what is appearing in the db,

    {
	    "_id" : ObjectId("..."),
	    "delete_flag" : false,
	    "socket" : "....",
	    "serial_number" : "OP000108",
	    "local_ip" : "10.30.0.94",
	    "mac_address" : "74:DA:38:2B:18:6D",
	    "public_ip" : "66.159.242.14",
	    "printer_id" : ObjectId("..."),
	    "last_modified" : ISODate("2017-06-25T19:50:18.281Z"),
	    "__v" : 0,
	    "data" : {
	    	    "status" : 0
	    },
	    "gcode_commands" : [ ],
	    "rotate_image" : false
    }

[image]

[image]

markwal, 2017-06-27:
Hmmm...  my GPX printer sends 13 ok.   But it takes longer than I expect for it to show the connect button.  It does send a status, then a getUrl and the getUrl, I think doesn't have a status in it.  Maybe that nukes it to 0 until the next status?

dnewman-polar3d, 2017-06-27:
The getUrl will not impact the status.  Indeed, it causes no update of any form to the database.  HOWEVER, I've seen some cases where the status update was sent right on the heels of the hello.  As the hello had not completed processing, the status update was ignored, the socket not yet being validated but the connection established less than 60s in the past (and thus within the grace period).

dnewman-polar3d, 2017-06-27:
Indeed, regarding the cases of a status being sent on the heels of a hello: I'd delay a few seconds (e.g., 5s) between sending the hello and then sending anything else.  (5s is likely more than needed.)  Point being that the hello involves a db lookup and since this is Node.js, it'll get put aside while other commands are accepted.  If a status comes in too soon, before the hello command has completed all of its processing, that status command is summarily ignored.

dnewman-polar3d, 2017-06-27:
I can now confirm that on the plugin starting up, I see in the octoprint log

1. status 13 sent @ 15:03:42,656
2. hello sent @ 15:03:42,709

So the status was sent **before** the hello!  At any rate, it was ignored by the cloud and so the status was not updated.  What shows in the cloud is whatever was still in the db from the last connection that successfully sent a status (after validation).

Then eventually a status 13 is again sent so the CONNECT button does eventually appear.

If there is an item in the printer's queue, the CONNECT button works.  Owing to a bug in the web server's SPA code, if there is nothing queued, clicking CONNECT does not work (error displayed in javascript console).  This is fixed and will be deployed this evening.

NOW, I notice that when I click CONNECT, the printer receives it.  But its next status update is a 12.  That causes the polar cloud to think there's an error.  But then the next status from the plugin is then a 0 which means all is A-Okay and the printer is idle.

markwal, 2017-06-30:
I believe this is fixed by dd1446e6

closed 2017-06-30 by markwal