markwal/OctoPrint-PolarCloud #8: Camer snapshots uploading as 0 byte files?

dnewman-polar3d, 2017-06-25:
This is a printer with the GPX plugin.

I can see in S3 that the file timestamps are changing -- both idle and printing images -- but the files are 0 bytes long.

markwal, 2017-06-25:
Is this across the board?  IE is anybody successfully uploading non-0 byte files at the moment?

dnewman-polar3d, 2017-06-25:
No... Without looking at the images, I can see that 107 uploads 30K or so sized images.  However 106 is also uploading 0 byte images.

PB000108 has an RPi camera.... Not a USB camera.  If there's something to look for in the logs, let me know.  Was going to do that but lunch is getting cold.

dnewman-polar3d, 2017-06-25:
2017-06-25 20:06:12,253 - octoprint.plugins.polarcloud - DEBUG - _upload_snapshot
2017-06-25 20:06:12,255 - octoprint.plugins.polarcloud - DEBUG - upload_type printing
2017-06-25 20:06:14,199 - octoprint.plugins.polarcloud - DEBUG - 204: 
2017-06-25 20:06:14,201 - octoprint.plugins.polarcloud - DEBUG - Image captured from http://127.0.0.1:8080/?action=snapshot

dnewman-polar3d, 2017-06-25:
I don't see anything that matches .jpeg, .jpg, or .png down in .octoprint either.  (E.g., find .octoprint -name \*.jpg turns up nada.)

markwal, 2017-06-25:
Octoprint only stores snapshots itself while recording timelapse.  The plugin retrieves and posts entirely from RAM (doesn't create a file).

dnewman-polar3d, 2017-06-26:
Here's a data point (or 2) for you.  I have one Octopi which uploads 0 byte images.  I have a second which uploads ~30K images.  The difference?  The ~30K images are with a USB camera.  The 0 byte images are with an Raspberry Pi camera.

markwal, 2017-06-26:
Hmm..  I had one of each for my testing as well.  If you update to the latest plugin and turn on verbose logging, it should log the URL that is producing 0 byte files. If you visit that url in the browser, do you see the snapshot?

dnewman-polar3d, 2017-06-26:
No clue how to do that.... update a plugin installed by Octoprint....

dnewman-polar3d, 2017-06-26:
Something similar to this.... (but with no need to submodule I suspect.)

    cd ~
    git clone https://github.com/markwal/OctoPrint-GPX
    cd OctoPrint-GPX
    git submodule update --init
    source ~/oprint/bin/activate
    python setup.py develop
    sudo service octoprint restart

That's from Sep 30, 2015...

dnewman-polar3d, 2017-06-26:
Installed the latest but there seems to be a coding error...?

    2017-06-26 04:49:20,122 - octoprint.plugins.polarcloud - DEBUG - getUrlResponse {u'status': u'SUCCESS', u'expires': 86400, u'url': u'https://s3.amazonaws.com/polar3d.com', u'fields': {u'X-Amz-Algorithm': u'AWS4-HMAC-SHA256', u'key': u'files/printer/OP000108/30wUDVtXoCKnCwpC67HzqCKe-current.jpg', u'bucket': u'polar3d.com', u'acl': u'public-read', u'X-Amz-Signature': u'da399a925cf53e8bafe38daec4fe220ffd', u'X-Amz-Date': u'20170626T044919Z', u'Policy': u'eyJleHBpcmF0aW9uIj6QA/20170626/us-east-1/s3/aws4_request'}, u'serialNumber': u'OP000108', u'ctype': u'image/jpeg', u'maxSize': 153600, u'type': u'idle', u'method': u'post'}
    2017-06-26 04:49:20,202 - octoprint.plugins.polarcloud - DEBUG - response_type = idle
    2017-06-26 04:49:21,295 - octoprint.plugins.polarcloud - DEBUG - _upload_snapshot
    2017-06-26 04:49:21,296 - octoprint.plugins.polarcloud - DEBUG - upload_type idle
    2017-06-26 04:49:21,338 - octoprint.plugins.polarcloud - DEBUG - Recompressing snapshot to smaller size
    2017-06-26 04:49:21,488 - octoprint.plugins.polarcloud - ERROR - Could not post snapshot to PolarCloud
    Traceback (most recent call last):
      File "/home/pi/OctoPrint-PolarCloud/octoprint_polarcloud/__init__.py", line 494, in _upload_snapshot
        if len(image_bytes) == 0:
    AttributeError: StringIO instance has no attribute '__len__'
    2017-06-26 04:49:22,633 - octoprint.server.preemptive_cache - INFO - ... done in 33.92s

dnewman-polar3d, 2017-06-26:
2017-06-26 14:34:15,760 - octoprint.plugins.polarcloud - DEBUG - response_type = idle
    2017-06-26 14:34:16,961 - octoprint.plugins.polarcloud - DEBUG - _upload_snapshot
    2017-06-26 14:34:17,021 - octoprint.plugins.polarcloud - DEBUG - upload_type idle
    2017-06-26 14:34:17,423 - octoprint.plugins.polarcloud - DEBUG - Recompressing snapshot to smaller size
    2017-06-26 14:34:17,905 - octoprint.plugins.polarcloud - DEBUG - Image transcoded from size 198444 to 35963
    2017-06-26 14:34:18,775 - octoprint.server.preemptive_cache - INFO - ... done in 3.70s
    2017-06-26 14:34:19,824 - octoprint.plugins.polarcloud - DEBUG - 204: 
    2017-06-26 14:34:19,827 - octoprint.plugins.polarcloud - DEBUG - Image captured from http://127.0.0.1:8080/?action=snapshot

If I manually get a snapshot,

     curl -o foo.jpg http://127.0.0.1:8080/?action=snapshot
      % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                     Dload  Upload   Total   Spent    Left  Speed
    100  194k    0  194k    0     0  2061k      0 --:--:-- --:--:-- --:--:-- 2088k

So I did get a snapshot manually (and it appears that the plugin did as well).  Perhaps something off with the handling of the compression?

markwal, 2017-06-26:
The size debug log was one of the new ones I added.  I was hoping it would say 0 as the output size and then we'd know the recompression was failing (semi-silently).

Now that I fixed measuring the length of a StringIO, that shows that even if the compression didn't produce a working image it is not zero len.  Must be an upload failure.  I was wondering if maybe the compression might leave the seek pointer in a random spot.  So now it is asserted at 0 after seeking to the end to discover the length.

dnewman-polar3d, 2017-06-26:
It works now....  Seeing ~30K images uploaded

closed 2017-06-26 by dnewman-polar3d