markwal/OctoPrint-PolarCloud #10: Installation Errors (Could not parse output from pip)

nyco007, 2017-06-26:
Installing plugin "PolarCloud (Beta)" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
/usr/bin/python2 -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip --user
Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip (2.9MB)
Requirement already satisfied: OctoPrint in /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.4-py2.7.egg (from OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: SocketIO-client in /usr/local/lib/python2.7/dist-packages (from OctoPrint-PolarCloud==0.1.0Beta)
Collecting pyopenssl>=16.1 (from OctoPrint-PolarCloud==0.1.0Beta)
Using cached pyOpenSSL-17.0.0-py2.py3-none-any.whl
Collecting Pillow (from OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: flask<0.11,>=0.9 in /usr/lib/python2.7/dist-packages (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: Jinja2<2.9,>=2.8 in /usr/local/lib/python2.7/dist-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /usr/local/lib/python2.7/dist-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: tornado==4.0.2 in /usr/local/lib/python2.7/dist-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: sockjs-tornado<1.1,>=1.0.2 in /usr/local/lib/python2.7/dist-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /usr/local/lib/python2.7/dist-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /usr/local/lib/python2.7/dist-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /usr/local/lib/python2.7/dist-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /usr/local/lib/python2.7/dist-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /usr/local/lib/python2.7/dist-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: markdown<2.7,>=2.6.4 in /usr/local/lib/python2.7/dist-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: pyserial<2.8,>=2.7 in /usr/local/lib/python2.7/dist-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: netaddr<0.8,>=0.7.17 in /usr/local/lib/python2.7/dist-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /usr/local/lib/python2.7/dist-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: sarge<0.2,>=0.1.4 in /usr/local/lib/python2.7/dist-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: netifaces<0.11,>=0.10 in /usr/local/lib/python2.7/dist-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /usr/local/lib/python2.7/dist-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: rsa<3.3,>=3.2 in /usr/local/lib/python2.7/dist-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /usr/local/lib/python2.7/dist-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: requests<2.8,>=2.7 in /usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /usr/local/lib/python2.7/dist-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: psutil<3.3,>=3.2.1 in /usr/local/lib/python2.7/dist-packages/psutil-3.2.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: Click<6.3,>=6.2 in /usr/local/lib/python2.7/dist-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /usr/local/lib/python2.7/dist-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /usr/local/lib/python2.7/dist-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /usr/local/lib/python2.7/dist-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: future<0.16,>=0.15 in /usr/local/lib/python2.7/dist-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: scandir<1.4,>=1.3 in /usr/local/lib/python2.7/dist-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: websocket-client<0.41,>=0.40 in /usr/local/lib/python2.7/dist-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: python-dateutil<2.7,>=2.6 in /usr/local/lib/python2.7/dist-packages/python_dateutil-2.6.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from SocketIO-client->OctoPrint-PolarCloud==0.1.0Beta)
Collecting cryptography>=1.7 (from pyopenssl>=16.1->OctoPrint-PolarCloud==0.1.0Beta)
Using cached cryptography-1.9.tar.gz
Requirement already satisfied: olefile in /usr/local/lib/python2.7/dist-packages (from Pillow->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: itsdangerous>=0.21 in /usr/lib/python2.7/dist-packages (from flask<0.11,>=0.9->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: MarkupSafe in /usr/lib/python2.7/dist-packages (from Jinja2<2.9,>=2.8->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: certifi in /usr/local/lib/python2.7/dist-packages (from tornado==4.0.2->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: backports.ssl_match_hostname in /usr/local/lib/python2.7/dist-packages/backports.ssl_match_hostname-3.5.0.1-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: blinker in /usr/lib/python2.7/dist-packages (from Flask-Principal<0.4,>=0.3.5->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: Babel>=1.0 in /usr/local/lib/python2.7/dist-packages/Babel-2.4.0-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: speaklater>=1.2 in /usr/local/lib/python2.7/dist-packages/speaklater-1.3-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: webassets>=0.10 in /usr/local/lib/python2.7/dist-packages/webassets-0.12.1-py2.7.egg (from Flask-Assets<0.11,>=0.10->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: argh>=0.24.1 in /usr/local/lib/python2.7/dist-packages/argh-0.26.2-py2.7.egg (from watchdog<0.9,>=0.8.3->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: pathtools>=0.1.1 in /usr/local/lib/python2.7/dist-packages/pathtools-0.1.2-py2.7.egg (from watchdog<0.9,>=0.8.3->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: pyasn1>=0.1.3 in /usr/lib/python2.7/dist-packages (from rsa<3.3,>=3.2->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
Requirement already satisfied: regex in /usr/local/lib/python2.7/dist-packages/regex-2017.06.23-py2.7-linux-armv7l.egg (from awesome-slugify<1.7,>=1.6.5->OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
[... cut ...]
Command "/usr/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-AglOjN/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-dpIn9B-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-AglOjN/cryptography/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

markwal, 2017-06-26:
What OS are you installing on?  It looks like you're not running OctoPrint from a virtual environment.  If you do "pip freeze" from the command line in your OctoPrint folder, what version of pyopenssl, openssl and cryptography modules do you have installed?

nyco007, 2017-06-26:
Im runing debian jessy... I'm really new to that stuff, but here is what pip freeze brought up:

Adafruit-BBIO==1.0.3
argh==0.26.2
asn1crypto==0.22.0
astroid==1.2.1
awesome-slugify==1.6.5
Babel==2.4.0
backports.ssl-match-hostname==3.5.0.1
blinker==1.3
certifi==2017.4.17
cffi==0.8.6
chainmap==1.0.2
chardet==3.0.4
click==6.2
colorama==0.3.2
cryptography==0.6.1
enum34==1.1.6
feedparser==5.2.1
Flask==0.10.1
Flask-Assets==0.10
Flask-Babel==0.9
Flask-Login==0.2.11
Flask-Principal==0.3.5
future==0.15.2
html5lib==0.999
idna==2.5
ipaddress==1.0.18
itsdangerous==0.24
iw-parse==0.0.2
Jinja2==2.8.1
logilab-common==0.62.0
Markdown==2.6.8
MarkupSafe==0.23
ndg-httpsclient==0.3.2
netaddr==0.7.19
netifaces==0.10.6
OctoPrint==1.3.4
OctoPrint-Fullscreen==0.0.3
olefile==0.44
pathtools==0.1.2
pkginfo==1.2.1
ply==3.4
psutil==3.2.2
pyasn1==0.1.7
PyBBIO==0.10
pycparser==2.10
pygobject==3.14.0
pyinotify==0.9.4
pylint==1.3.1
pylru==1.0.9
pyOpenSSL==0.14
pyserial==2.7
python-dateutil==2.6.0
pytz==2017.2
PyYAML==3.10
regex==2017.6.23
requests==2.7.0
rsa==3.2.3
sarge==0.1.4
scandir==1.3
semantic-version==2.4.2
serbus==1.0.5
six==1.8.0
socketIO-client==0.7.2
sockjs-tornado==1.0.3
speaklater==1.3
tornado==4.0.2
Unidecode==0.4.20
urllib3==1.21.1
virtualenv==15.1.0
watchdog==0.8.3
webassets==0.12.1
websocket-client==0.40.0
Werkzeug==0.8.3

markwal, 2017-06-27:
This is due to the following:
https://github.com/pyca/cryptography/issues/3392

You should be able to fix it by manually installing cffi and then trying pyOpenSSL again.

pip install --upgrade cffi
pip install --upgrade pyOpenSSL

closed 2017-06-27 by markwal

nyco007, 2017-06-27:
pip install --upgrade pyOpenSSL returns some errors...:

nyco007@beaglebone:~$ sudo pip install --upgrade pyOpenSSL
Collecting pyOpenSSL
  Using cached pyOpenSSL-17.0.0-py2.py3-none-any.whl
Requirement already up-to-date: six>=1.5.2 in /usr/local/lib/python2.7/dist-packages (from pyOpenSSL)
Collecting cryptography>=1.7 (from pyOpenSSL)
  Using cached cryptography-1.9.tar.gz
Requirement already up-to-date: idna>=2.1 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: asn1crypto>=0.21.0 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: enum34 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: ipaddress in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: cffi>=1.7 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: pycparser in /usr/local/lib/python2.7/dist-packages (from cffi>=1.7->cryptography>=1.7->pyOpenSSL)
Building wheels for collected packages: cryptography
  Running setup.py bdist_wheel for cryptography ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Svc5pz/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpcIi2kapip-wheel- --python-tag cp27:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-Svc5pz/cryptography/setup.py", line 321, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
      _setup_distribution = dist = klass(attrs)
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 267, in __init__
      _Distribution.__init__(self,attrs)
    File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
      self.finalize_options()
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 302, in finalize_options
      ep.load()(self, ep.name, value)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 188, in cffi_modules
      add_cffi_module(dist, cffi_module)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 49, in add_cffi_module
      execfile(build_file_name, mod_vars)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 25, in execfile
      exec(code, glob, glob)
    File "src/_cffi_src/build_openssl.py", line 82, in <module>
      extra_link_args=extra_link_args(compiler_type()),
    File "/tmp/pip-build-Svc5pz/cryptography/src/_cffi_src/utils.py", line 61, in build_ffi_for_binding
      extra_link_args=extra_link_args,
    File "/tmp/pip-build-Svc5pz/cryptography/src/_cffi_src/utils.py", line 69, in build_ffi
      ffi = FFI()
    File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 54, in __init__
      backend.__version__, backend.__file__))
  Exception: Version mismatch: this is the 'cffi' package version 1.10.0, located in '/usr/local/lib/python2.7/dist-packages/cffi/api.pyc'.  When we import the top-level '_cffi_backend' extension module, we get version 0.8.6, located in '/usr/lib/python2.7/dist-packages/_cffi_backend.arm-linux-gnueabihf.so'.  The two versions should be equal; check your installation.

  ----------------------------------------
  Failed building wheel for cryptography
  Running setup.py clean for cryptography
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Svc5pz/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-Svc5pz/cryptography/setup.py", line 321, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
      _setup_distribution = dist = klass(attrs)
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 267, in __init__
      _Distribution.__init__(self,attrs)
    File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
      self.finalize_options()
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 302, in finalize_options
      ep.load()(self, ep.name, value)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 188, in cffi_modules
      add_cffi_module(dist, cffi_module)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 49, in add_cffi_module
      execfile(build_file_name, mod_vars)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 25, in execfile
      exec(code, glob, glob)
    File "src/_cffi_src/build_openssl.py", line 82, in <module>
      extra_link_args=extra_link_args(compiler_type()),
    File "/tmp/pip-build-Svc5pz/cryptography/src/_cffi_src/utils.py", line 61, in build_ffi_for_binding
[... 22 lines cut ...]
      File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 267, in __init__
        _Distribution.__init__(self,attrs)
      File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
[... cut ...]
    ----------------------------------------
  Rolling back uninstall of cryptography
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Svc5pz/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-ZISvBH-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Svc5pz/cryptography/

markwal, 2017-06-27:
Ah oh.  Your cffi package got mangled.  You've got the python from one version and the built stuff from another.  Oof.  It'll probably take a few tries to get this fixed up as I can't replicate your state here.

How about:

sudo apt-get update
sudo apt-get install python-cffi python-openssl

nyco007, 2017-06-27:
Unfortunally, since pip install --upgrade cffi my OctoPrint is not starting anymore...

nyco007@beaglebone:~/OctoPrint$ ./run
Traceback (most recent call last):
  File "./run", line 22, in <module>
    octoprint.main()
  File "/home/nyco007/OctoPrint/src/octoprint/__init__.py", line 404, in main
    from octoprint.cli import octo
  File "/home/nyco007/OctoPrint/src/octoprint/cli/__init__.py", line 136, in <module>
    from .client import client_commands
  File "/home/nyco007/OctoPrint/src/octoprint/cli/client.py", line 10, in <module>
    import octoprint_client
  File "/home/nyco007/OctoPrint/src/octoprint_client/__init__.py", line 8, in <module>
    import requests
  File "/usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg/requests/__init__.py", line 53, in <module>
    from .packages.urllib3.contrib import pyopenssl
  File "/usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg/requests/packages/urllib3/contrib/pyopenssl.py", line 53, in <module>
    import OpenSSL.SSL
  File "/usr/lib/python2.7/dist-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import rand, crypto, SSL
  File "/usr/lib/python2.7/dist-packages/OpenSSL/rand.py", line 11, in <module>
    from OpenSSL._util import (
  File "/usr/lib/python2.7/dist-packages/OpenSSL/_util.py", line 4, in <module>
    binding = Binding()
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py", line 89, in __init__
    self._ensure_ffi_initialized()
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py", line 113, in _ensure_ffi_initialized
    libraries=libraries,
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/utils.py", line 39, in build_ffi
    ffi = cffi.FFI()
  File "/home/nyco007/.local/lib/python2.7/site-packages/cffi/api.py", line 54, in __init__
    backend.__version__, backend.__file__))
Exception: Version mismatch: this is the 'cffi' package version 1.10.0, located in '/home/nyco007/.local/lib/python2.7/site-packages/cffi/api.pyc'.  When we import the top-level '_cffi_backend' extension module, we get version 0.8.6, located in '/usr/lib/python2.7/dist-packages/_cffi_backend.arm-linux-gnueabihf.so'.  The two versions should be equal; check your installation.

nyco007, 2017-06-27:
I figured out to uninstall the cffi 1.10.0 with sudo apt-get remove python-cffi and tired then sudo apt-get install python-cffi python-openssl as you proposed.
Now the installation worked without issues. Thanks!

OptogeneticsandNeuralEngineeringCore, 2017-07-24:
Thanks a bunch, everyone. I followed these steps, but also ran into issue with my pip being messed up. So for others having issue, it may be cool to try: 
sudo apt-get update

Then pip check pip with pip freeze. If error:
sudo apt-get remove --auto-remove python-pip  
wget https://bootstrap.pypa.io/get-pip.py 
sudo python get-pip.py  
pip freeze

Then 
sudo apt-get remove python-cffi
sudo apt-get install python-cffi python-openssl

sudo reboot
???
Profit.

R0dri, 2017-07-25:
I had the same error output caused by that ciff package. I did all things mentioned above and more that I don't even recall. In the end, I manage to fix by removing ciff by an "apt-get remove python-cffi" and by "pip uninstall cffi". When I installed it as described above and then had no problems with the "cryptography" package as the error posted also above.

But now I still get the same "Could Not parse output from pip" error. Lookin at the log I found this:
> Running setup.py install for Pillow: started
Command "/home/pi/OctoPrint/venv/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-s8d6RN/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-MMy_K0-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/OctoPrint/venv/include/site/python2.7/Pillow" failed with error code 1 in /tmp/pip-build-s8d6RN/Pillow/
Running setup.py install for Pillow: finished with status 'error'

I will try to find anything related to this particular error, any thoughts? i will keep searching a little further and post any update.

I am running Jessie.

R0dri, 2017-07-25:
I figured it out! Searching a little further the log I found this message:

>  reading manifest file 'Pillow.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.sh'
    no previously-included directories found matching 'docs/_static'
    warning: no previously-included files found matching '.coveragerc'
    warning: no previously-included files found matching '.editorconfig'
    warning: no previously-included files found matching '.landscape.yaml'
    warning: no previously-included files found matching '.travis'
    warning: no previously-included files found matching '.travis/*'
    warning: no previously-included files found matching 'appveyor.yml'
    warning: no previously-included files found matching 'build_children.sh'
    warning: no previously-included files found matching 'tox.ini'
    warning: no previously-included files matching '.git*' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '*.so' found anywhere in distribution
    writing manifest file 'Pillow.egg-info/SOURCES.txt'
    running build_ext
    
    
    The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source.
    
    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-snbzYZ/pillow/setup.py", line 788, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:
    
    The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source.
    
    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html

I found this, https://github.com/python-pillow/Pillow/issues/2163

It seems that I was missing `libjpeg` dependency. I installed it with:
`sudo apt-get install libjpeg-dev`
Finally:
`sudo pip install pillow`

I hope this helps. Should I post this as a new error? Thanks

samuelcarreira, 2018-02-20:
@OptogeneticsandNeuralEngineeringCore your solution works for me on a orange pi zero