markwal/OctoPrint-PolarCloud #11: Timelapse not yet uploaded

markwal, 2017-06-27:


markwal, 2017-07-01:
I believe the feature/timelapse branch may have this working now.  Testing.

markwal, 2017-07-04:
Fixed in 0.2.0.  After doing some more extensibility work in OctoPrint, need to revisit to: only generate timelapse once.  Currently OctoPrint generates one in mpg format and then the PolarCloud plugin transcodes it into mp4 format.  Also, Polar Cloud would like a normalized length of time lapses.

closed 2017-07-04 by markwal

dnewman-polar3d, 2017-07-04:
BTW and FYI: Polar Software keeps the .mp4 file about along with serial number and job Id and then assumes it knows where the file lives in S3.  With that info, some assumed, it then sees if the .mp4 actually got uploaded to S3 (e.g., user didn't turn the printer off).  If it did get uploaded (HTTP HEAD matches), then it deletes it.  Otherwise, it attempts to upload it again.

Should the plugin do that?  Maybe if people report issues.  Just wanted to mention this for the record.

markwal, 2017-07-04:
That's a good idea.  Right now, the plugin will only make one attempt. And it keeps the local copy indefinitely. It'll show up in the OctoPrint UI as a timelapse also.