markwal/OctoPrint-PolarCloud #12: Should add a message mechanism with helpers for some config issues

markwal, 2017-06-27:
Like:

- [ ] Couldn't create a public key
- [ ] Email+Pin not correct

dnewman-polar3d, 2017-06-27:
I can update the servers to at least indicate an issue with the email and/or PIN.  It'll subsume the following cases:
1. email address as supplied does not match any existing account.  (It's all converted to lower case so it's not an upper/lower case mismatch issue).
2. the PIN does not match the PIN of the indicated account.

dnewman-polar3d, 2017-06-27:
Errors returned as "reason" field
1. mfg field missing: MFG_MISSING (plugin error)
2. mfg field not recognized: MFG_UNKNWON
3. No user matching email: EMAIL_PIN_ERROR
4. User account lacking a PIN (should never happen): EMAIL_PIN_ERROR
5. PINs don't match (email matched an account, but PINs didn't match): EMAIL_PIN_ERROR
6. Printer already exists, email & PIN match, account already owns the printer: SUCCESS (status SUCCESS as well)
7. Printer already exists, email & PIN match, account does NOT own the printer (MAC address): FORBIDDEN
8. Db error generating a new serial number: SERVER_ERROR
9. All good: SUCCESS (status SUCCESS as well)
10. More db error cases: SERVER_ERROR

dnewman-polar3d, 2017-06-27:
development server is now refreshed with the above changes; I'll update protocol spec next.

markwal, 2017-07-02:
I think latest master has all the ones listed here.

closed 2017-07-02 by markwal