markwal/OctoPrint-PolarCloud #13: Printer Status Offline

nyco007, 2017-06-27:
I registered OctoPrint to Polar Cloud and the Printer also shows up in Polar Cloud but its status is offline.
I also recognized some failure reports on the command line of OctoPrint:

2017-06-28 00:39:38,827 - octoprint.plugins.polarcloud - ERROR - polar_heartbeat exception
Traceback (most recent call last):
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 460, in _polar_status_heartbeat
    self._socket.wait(seconds=1)
  File "/usr/local/lib/python2.7/dist-packages/socketIO_client/__init__.py", line 251, in wait
    self._process_packets()
  File "/usr/local/lib/python2.7/dist-packages/socketIO_client/__init__.py", line 276, in _process_packets
    for engineIO_packet in self._transport.recv_packet():
  File "/usr/local/lib/python2.7/dist-packages/socketIO_client/transports.py", line 158, in recv_packet
    packet_text)
  File "/usr/local/lib/python2.7/dist-packages/socketIO_client/parsers.py", line 96, in parse_packet_text
    packet_type = int(get_character(packet_text, 0))
  File "/usr/local/lib/python2.7/dist-packages/socketIO_client/symmetries.py", line 33, in get_character
    return chr(get_byte(x, index))
  File "/usr/local/lib/python2.7/dist-packages/socketIO_client/symmetries.py", line 29, in get_byte
    return indexbytes(x, index)
  File "/usr/local/lib/python2.7/dist-packages/six.py", line 655, in indexbytes
    return ord(buf[i])
IndexError: string index out of range
2017-06-28 00:39:43,860 - octoprint.plugins.polarcloud - INFO - Socket disconnected, clear and restart
2017-06-28 00:39:43,871 - octoprint.plugins.polarcloud - WARNING - unable to create socket to Polar Cloud, check again in 2.79280137592 seconds
2017-06-28 00:39:49,987 - octoprint.plugins.polarcloud - ERROR - polar_heartbeat exception
Traceback (most recent call last):
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 455, in _polar_status_heartbeat
    task()
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 594, in _hello
    'signature': base64.b64encode(crypto.sign(self._key, self._challenge, b'sha256')),
  File "/usr/lib/python2.7/dist-packages/OpenSSL/crypto.py", line 2092, in sign
    _lib.EVP_SignUpdate(md_ctx, data, len(data))
TypeError: initializer for ctype 'void *' must be a cdata pointer, not unicode

markwal, 2017-06-28:
The first exception happens when Polar hangs up on the socket (it can do that if we don't authenticate within a certain period for example).

The second is more concerning because it's apparently an incompatibility with the version of OpenSSL you have (or a bug).  Could you do a "pip freeze" with your virtual env activated and let me know what version of pyOpenSSL you have?

nyco007, 2017-06-28:
Hi markwal,

here is the result of pip freeze:

nyco007@beaglebone:~$ pip freeze
Adafruit-BBIO==1.0.3
argh==0.26.2
asn1crypto==0.22.0
astroid==1.2.1
awesome-slugify==1.6.5
Babel==2.4.0
backports.ssl-match-hostname==3.5.0.1
blinker==1.3
certifi==2017.4.17
cffi==0.8.6
chainmap==1.0.2
chardet==3.0.4
click==6.2
colorama==0.3.2
cryptography==0.6.1
enum34==1.1.6
feedparser==5.2.1
Flask==0.10.1
Flask-Assets==0.10
Flask-Babel==0.9
Flask-Login==0.2.11
Flask-Principal==0.3.5
future==0.15.2
html5lib==0.999
idna==2.5
ipaddress==1.0.18
itsdangerous==0.24
iw-parse==0.0.2
Jinja2==2.8.1
logilab-common==0.62.0
Markdown==2.6.8
MarkupSafe==0.23
netaddr==0.7.19
netifaces==0.10.6
OctoPrint==1.3.4
OctoPrint-Fullscreen==0.0.3
OctoPrint-PolarCloud==0.1.0b0
olefile==0.44
pathtools==0.1.2
Pillow==4.1.1
pkginfo==1.2.1
ply==3.4
psutil==3.2.2
pyasn1==0.1.7
PyBBIO==0.10
pycparser==2.17
pygobject==3.14.0
pyinotify==0.9.4
pylint==1.3.1
pylru==1.0.9
pyOpenSSL==0.14
pyserial==2.7
python-dateutil==2.6.0
pytz==2017.2
PyYAML==3.10
regex==2017.6.23
requests==2.7.0
rsa==3.2.3
sarge==0.1.4
scandir==1.3
semantic-version==2.4.2
serbus==1.0.5
six==1.10.0
socketIO-client==0.7.2
sockjs-tornado==1.0.3
speaklater==1.3
tornado==4.0.2
Unidecode==0.4.20
urllib3==1.21.1
virtualenv==15.1.0
watchdog==0.8.3
webassets==0.12.1
websocket-client==0.40.0
Werkzeug==0.8.3

markwal, 2017-06-29:
Hmmm...  yes, pyOpenSSL 0.14 is likely too old.  I removed the minimum version check because it caused problems with the particular OpenSSL installed in octopi 0.13 and removed the dependency that was causing problems with its version of OpenSSL (dump_publickey)

Are you able to upgrade OpenSSL:

pip install --upgrade pyOpenSSL

?

nyco007, 2017-06-29:
Unfortunaly no... The Polar Plugin also is not shown anymore at OctoPrint plugins. If i try to reinstall, it work, but it remain not showing up.... but here is re result of trying to upgrade pyOpenSSL:

nyco007@beaglebone:~/OctoPrint$ sudo pip install --upgrade pyOpenSSL
Collecting pyOpenSSL
  Using cached pyOpenSSL-17.0.0-py2.py3-none-any.whl
Requirement already up-to-date: six>=1.5.2 in /usr/local/lib/python2.7/dist-packages (from pyOpenSSL)
Collecting cryptography>=1.7 (from pyOpenSSL)
  Using cached cryptography-1.9.tar.gz
Requirement already up-to-date: idna>=2.1 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: asn1crypto>=0.21.0 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: enum34 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: ipaddress in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: cffi>=1.7 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: pycparser in /usr/local/lib/python2.7/dist-packages (from cffi>=1.7->cryptography>=1.7->pyOpenSSL)
Building wheels for collected packages: cryptography
  Running setup.py bdist_wheel for cryptography ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-nQ3Y3r/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpwmIEASpip-wheel- --python-tag cp27:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-nQ3Y3r/cryptography/setup.py", line 321, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
      _setup_distribution = dist = klass(attrs)
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 267, in __init__
      _Distribution.__init__(self,attrs)
    File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
      self.finalize_options()
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 302, in finalize_options
      ep.load()(self, ep.name, value)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 188, in cffi_modules
      add_cffi_module(dist, cffi_module)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 49, in add_cffi_module
      execfile(build_file_name, mod_vars)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 25, in execfile
      exec(code, glob, glob)
    File "src/_cffi_src/build_openssl.py", line 82, in <module>
      extra_link_args=extra_link_args(compiler_type()),
    File "/tmp/pip-build-nQ3Y3r/cryptography/src/_cffi_src/utils.py", line 61, in build_ffi_for_binding
      extra_link_args=extra_link_args,
    File "/tmp/pip-build-nQ3Y3r/cryptography/src/_cffi_src/utils.py", line 69, in build_ffi
      ffi = FFI()
    File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 54, in __init__
      backend.__version__, backend.__file__))
  Exception: Version mismatch: this is the 'cffi' package version 1.10.0, located in '/usr/local/lib/python2.7/dist-packages/cffi/api.pyc'.  When we import the top-level '_cffi_backend' extension module, we get version 0.8.6, located in '/usr/lib/python2.7/dist-packages/_cffi_backend.arm-linux-gnueabihf.so'.  The two versions should be equal; check your installation.

  ----------------------------------------
  Failed building wheel for cryptography
  Running setup.py clean for cryptography
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-nQ3Y3r/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-nQ3Y3r/cryptography/setup.py", line 321, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
      _setup_distribution = dist = klass(attrs)
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 267, in __init__
      _Distribution.__init__(self,attrs)
    File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
      self.finalize_options()
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 302, in finalize_options
      ep.load()(self, ep.name, value)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 188, in cffi_modules
      add_cffi_module(dist, cffi_module)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 49, in add_cffi_module
      execfile(build_file_name, mod_vars)
    File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 25, in execfile
      exec(code, glob, glob)
    File "src/_cffi_src/build_openssl.py", line 82, in <module>
      extra_link_args=extra_link_args(compiler_type()),
    File "/tmp/pip-build-nQ3Y3r/cryptography/src/_cffi_src/utils.py", line 61, in build_ffi_for_binding
[... 22 lines cut ...]
[... cut ...]
    ----------------------------------------
  Rolling back uninstall of cryptography
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-nQ3Y3r/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-Qj3X_4-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-nQ3Y3r/cryptography/

markwal, 2017-06-29:
Oh man, the debian packages are too old and the pypi packages don't build, sigh...

It looks like from the logs that the uninstall of cffi that you mentioned in the other issue didn't remove cffi completely (version mismatch message above, it's still picking up 0.8.6)

You may want to do the uninstall thing again and then delete that file: /usr/lib/python2.7/dist-packages/_cffi_backend.arm-linux-gnueabihf.so

And then reinstall using apt-get:

sudo apt-get install python-cffi python-cryptography python-openssl

I think at that point, you'll still be getting the original message (but maybe that's also because it's finding the older cffi?)  If so, then you might try to do the pip install --upgrade pyOpenSSL and see if it fails in a different place (instead of choking on the cffi version mismatch).

nyco007, 2017-06-29:
Okay, i uninstalled cffi again (sudo apt-get remove python-cffi)
, checked if the file /usr/lib/python2.7/dist-packages/_cffi_backend.arm-linux-gnueabihf.so is there and it wasn't... so i did a pip freeze and ironical there is also showing up cffi==1.10.0..? Is it possible that I've got two versions of cffi installed? If yes, how do I deinstall the other version?

However I reinstalled using sudo apt-get install python-cffi python-cryptography python-openssl

Now the Plugin shows up again in Octoprint UI.

Nevertheless doing  pip install --upgrade pyOpenSSL shows the same error or it seems like this:

nyco007@beaglebone:~/OctoPrint$ sudo  pip install --upgrade pyOpenSSL
Collecting pyOpenSSL
  Using cached pyOpenSSL-17.0.0-py2.py3-none-any.whl
Requirement already up-to-date: six>=1.5.2 in /usr/local/lib/python2.7/dist-packages (from pyOpenSSL)
Collecting cryptography>=1.7 (from pyOpenSSL)
  Using cached cryptography-1.9.tar.gz
Requirement already up-to-date: idna>=2.1 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: asn1crypto>=0.21.0 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: enum34 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: ipaddress in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.7->pyOpenSSL)
Collecting cffi>=1.7 (from cryptography>=1.7->pyOpenSSL)
Requirement already up-to-date: pycparser in /usr/local/lib/python2.7/dist-packages (from cffi>=1.7->cryptography>=1.7->pyOpenSSL)
Building wheels for collected packages: cryptography
  Running setup.py bdist_wheel for cryptography ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Rkqz5C/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpckDb2fpip-wheel- --python-tag cp27:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-Rkqz5C/cryptography/setup.py", line 321, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
      _setup_distribution = dist = klass(attrs)
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 267, in __init__
      _Distribution.__init__(self,attrs)
    File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
      self.finalize_options()
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 302, in finalize_options
      ep.load()(self, ep.name, value)
    File "/usr/local/lib/python2.7/dist-packages/pkg_resources.py", line 2184, in load
      ['__name__'])
  ImportError: No module named setuptools_ext

  ----------------------------------------
  Failed building wheel for cryptography
  Running setup.py clean for cryptography
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Rkqz5C/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-Rkqz5C/cryptography/setup.py", line 321, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
      _setup_distribution = dist = klass(attrs)
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 267, in __init__
      _Distribution.__init__(self,attrs)
    File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
      self.finalize_options()
    File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 302, in finalize_options
      ep.load()(self, ep.name, value)
    File "/usr/local/lib/python2.7/dist-packages/pkg_resources.py", line 2184, in load
      ['__name__'])
  ImportError: No module named setuptools_ext

  ----------------------------------------
  Failed cleaning build dir for cryptography
Failed to build cryptography
Installing collected packages: cffi, cryptography, pyOpenSSL
  Found existing installation: cffi 0.8.6
    Uninstalling cffi-0.8.6:
      Successfully uninstalled cffi-0.8.6
  Found existing installation: cryptography 0.6.1
    Uninstalling cryptography-0.6.1:
      Successfully uninstalled cryptography-0.6.1
[... 8 lines cut ...]
      File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 267, in __init__
        _Distribution.__init__(self,attrs)
      File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
        self.finalize_options()
      File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 302, in finalize_options
        ep.load()(self, ep.name, value)
      File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 188, in cffi_modules
        add_cffi_module(dist, cffi_module)
      File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 49, in add_cffi_module
        execfile(build_file_name, mod_vars)
      File "/usr/local/lib/python2.7/dist-packages/cffi/setuptools_ext.py", line 25, in execfile
        exec(code, glob, glob)
      File "src/_cffi_src/build_openssl.py", line 82, in <module>
[... cut ...]
    ----------------------------------------
  Rolling back uninstall of cryptography
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Rkqz5C/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-4vXgfS-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Rkqz5C/cryptography/

markwal, 2017-06-30:
Hmmm...  So far, I'm stumped.  I think somehow the sequence is leaving you or giving you two different versions of cffi that can't coexist.

nyco007, 2017-07-01:
Hmm, can you recommend an actual guidance how to set up octoprint and mjpg streamer on a Beagle Bone Black? Then I would try setting up everything again from new. The guidance that i've found aren't actual anymore and are not working at every part, so i had to mix them up to get Octoprint and mjpg streamer working.

markwal, 2017-07-01:
I'll give it a whirl, but might take me a bit to get to it.

dnewman-polar3d, 2017-07-01:
I have some BBBs about.  (Gosh, one still has the old Atom OS that TI used to support.  Looks like a good candidate for an upgrade.)

dnewman-polar3d, 2017-07-01:
Using a newer BBB image I can get past cffi and pyOpenSSL issues but hit a problem with Pillow....
1. Download and flash to the BBB [bone-debian-8.7-lxqt-4gb-armhf-2017-03-19-4gb](https://debian.beagleboard.org/images/bone-debian-8.7-lxqt-4gb-armhf-2017-03-19-4gb.img.xz)
2. Followed directions at [Setup-BeagleBone-Black-Rev-C-(Jessie)](https://github.com/foosel/OctoPrint/wiki/Setup-BeagleBone-Black-Rev-C-(Jessie))
  a. Note that default account/pasword is now debian/temppwd
  b. Before installing Octoprint, I did
     sudo apt-get update
     sudo apt-get install build-essential python-dev python-setuptools python-pip python-smbus -y
    sudo pip install pyOpenSSL

I included python-smbus as, I seem to recall, it pulls in the cffi stuff.

I then proceeded with the Octoprint setup.  It went fine.  But when I then installed, via the Octoprint UI, the Polar Cloud plugin, there was difficulty with Pillow.  Lengthy log of the Polar Cloud plugin install follows.

    Requirement already satisfied: pyopenssl in /usr/lib/python2.7/dist-packages (from OctoPrint-PolarCloud==0.1.0Beta)
    
    Installing plugin "PolarCloud (Beta)" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
    /usr/bin/python2 -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip --user
    Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
    Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
    Requirement already satisfied: OctoPrint in /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.4-py2.7.egg (from OctoPrint-PolarCloud==0.1.0Beta)
    Collecting SocketIO-client (from OctoPrint-PolarCloud==0.1.0Beta)
    Downloading socketIO-client-0.7.2.tar.gz
    Requirement already satisfied: pyopenssl in /usr/lib/python2.7/dist-packages (from OctoPrint-PolarCloud==0.1.0Beta)
    Collecting Pillow (from OctoPrint-PolarCloud==0.1.0Beta)
    Downloading Pillow-4.2.0.tar.gz (12.7MB)
    Requirement already satisfied: flask<0.11,>=0.9 in /usr/lib/python2.7/dist-packages (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: Jinja2<2.9,>=2.8 in /usr/local/lib/python2.7/dist-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /usr/local/lib/python2.7/dist-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: tornado==4.0.2 in /usr/local/lib/python2.7/dist-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: sockjs-tornado<1.1,>=1.0.2 in /usr/local/lib/python2.7/dist-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: PyYAML<3.11,>=3.10 in /usr/local/lib/python2.7/dist-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /usr/local/lib/python2.7/dist-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /usr/local/lib/python2.7/dist-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /usr/local/lib/python2.7/dist-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /usr/local/lib/python2.7/dist-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: markdown<2.7,>=2.6.4 in /usr/local/lib/python2.7/dist-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: pyserial<2.8,>=2.7 in /usr/local/lib/python2.7/dist-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: netaddr<0.8,>=0.7.17 in /usr/local/lib/python2.7/dist-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: watchdog<0.9,>=0.8.3 in /usr/local/lib/python2.7/dist-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: sarge<0.2,>=0.1.4 in /usr/local/lib/python2.7/dist-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: netifaces<0.11,>=0.10 in /usr/local/lib/python2.7/dist-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: pylru<1.1,>=1.0.9 in /usr/local/lib/python2.7/dist-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: rsa<3.3,>=3.2 in /usr/local/lib/python2.7/dist-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /usr/local/lib/python2.7/dist-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
    Requirement already satisfied: requests<2.8,>=2.7 in /usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.1.0Beta)
[... cut ...]
    Command "/usr/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-lJib_J/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-Obbxhq-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-lJib_J/Pillow/
    Error!
    Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

dnewman-polar3d, 2017-07-01:
The Pillow failure may relate to my not doing

    sudo apt install gstreamer1.0-tools libx264-dev gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly

dnewman-polar3d, 2017-07-01:
Success... 

zlib headers or library files missing.... cannot build Pillow

    sudo apt-get install zlib1g-dev

After which

    The headers or library files could not be found for jpeg,

And thus

    sudo apt-get install zlib1g-dev libjpeg-dev python-libtiff libfreetype6-dev liblcms2-dev libwebp-dev tcl-dev tk-dev libopenjpeg-dev
    sudo pip install Pillow

After the above, I was then able to install the plugin and register with the Polar Cloud!

So, to summarize for a Beagle Bone Black

Download and flash to the BBB [bone-debian-8.7-lxqt-4gb-armhf-2017-03-19-4gb](https://debian.beagleboard.org/images/bone-debian-8.7-lxqt-4gb-armhf-2017-03-19-4gb.img.xz)

Note that the default account/password is debian/temppwd.

Before installing Octoprint and the Polar Cloud plugin, do

     sudo apt-get update
     sudo apt-get install build-essential python-dev python-setuptools python-pip \
            python-smbus gstreamer1.0-tools libx264-dev gstreamer1.0-plugins-good \
            gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly \
            zlib1g-dev libjpeg-dev python-libtiff libfreetype6-dev \
            liblcms2-dev libwebp-dev tcl-dev tk-dev libopenjpeg-dev
     sudo pip install pyOpenSSL
     sudo pip install Pillow

Follow the Octoprint for BBB directions at [Setup-BeagleBone-Black-Rev-C-(Jessie)](https://github.com/foosel/OctoPrint/wiki/Setup-BeagleBone-Black-Rev-C-(Jessie))

markwal, 2017-07-01:
Nice!  Thanks @dnewman-polar3d

nyco007, 2017-07-04:
Ok guys, firstly many thank for your help. I've just reflashed debian and installed OctoPrint, mjpg-streamer and even the installation of Polar Cloud went without problems!
BUT...I still have some problems, the printer keeps remaining offline after connecting it to Polar Cloud:

2017-07-04 01:17:05,318 - octoprint.plugins.polarcloud - INFO - Can't register because unable to communicate with Polar Cloud
2017-07-04 01:17:45,616 - octoprint.plugins.polarcloud - INFO - emit register
2017-07-04 01:17:47,390 - octoprint.plugins.polarcloud - WARNING - unable to create socket to Polar Cloud, check again in 1.95154827102 seconds
2017-07-04 01:17:53,541 - octoprint.plugins.polarcloud - ERROR - polar_heartbeat exception
Traceback (most recent call last):
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 432, in _wait_and_process
    task()
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 660, in _hello
    'signature': base64.b64encode(crypto.sign(self._key, self._challenge, b'sha256')),
  File "/usr/lib/python2.7/dist-packages/OpenSSL/crypto.py", line 2092, in sign
    _lib.EVP_SignUpdate(md_ctx, data, len(data))
TypeError: initializer for ctype 'void *' must be a cdata pointer, not unicode
2017-07-04 01:19:02,254 - octoprint.plugins.polarcloud - INFO - Socket disconnected, clear and restart
2017-07-04 01:19:02,264 - octoprint.plugins.polarcloud - WARNING - unable to create socket to Polar Cloud, check again in 1.66307886033 seconds
2017-07-04 01:19:07,307 - octoprint.plugins.polarcloud - ERROR - polar_heartbeat exception
Traceback (most recent call last):
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 432, in _wait_and_process
    task()
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 660, in _hello
    'signature': base64.b64encode(crypto.sign(self._key, self._challenge, b'sha256')),
  File "/usr/lib/python2.7/dist-packages/OpenSSL/crypto.py", line 2092, in sign
    _lib.EVP_SignUpdate(md_ctx, data, len(data))
TypeError: initializer for ctype 'void *' must be a cdata pointer, not unicode
2017-07-04 01:20:08,922 - octoprint.plugins.polarcloud - INFO - Socket disconnected, clear and restart
2017-07-04 01:20:08,928 - octoprint.plugins.polarcloud - WARNING - unable to create socket to Polar Cloud, check again in 2.55961905241 seconds
2017-07-04 01:20:14,919 - octoprint.plugins.polarcloud - ERROR - polar_heartbeat exception
Traceback (most recent call last):
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 432, in _wait_and_process
    task()
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 660, in _hello
    'signature': base64.b64encode(crypto.sign(self._key, self._challenge, b'sha256')),
  File "/usr/lib/python2.7/dist-packages/OpenSSL/crypto.py", line 2092, in sign
    _lib.EVP_SignUpdate(md_ctx, data, len(data))
TypeError: initializer for ctype 'void *' must be a cdata pointer, not unicode
2017-07-04 01:21:21,558 - octoprint.plugins.polarcloud - INFO - Socket disconnected, clear and restart
2017-07-04 01:21:21,567 - octoprint.plugins.polarcloud - WARNING - unable to create socket to Polar Cloud, check again in 1.60381588858 seconds
2017-07-04 01:21:26,786 - octoprint.plugins.polarcloud - ERROR - polar_heartbeat exception
Traceback (most recent call last):
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 432, in _wait_and_process
    task()
  File "/home/nyco007/.local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 660, in _hello
    'signature': base64.b64encode(crypto.sign(self._key, self._challenge, b'sha256')),
  File "/usr/lib/python2.7/dist-packages/OpenSSL/crypto.py", line 2092, in sign
    _lib.EVP_SignUpdate(md_ctx, data, len(data))
TypeError: initializer for ctype 'void *' must be a cdata pointer, not unicode

dnewman-polar3d, 2017-07-04:
Things are connecting to the Polar Cloud.  However, you still have issues with the crypto libraries and when the plugin attempts to use them, it gets an exception and the connection is then closed. Did you do all the steps I indicated in the order I indicated?  Before installing Octoprint there was a long list of other things to install first.  And you need to install Octoprint from github and build it (as opposed to finding a debian package with it already built).  That document I referenced earlier has those directions.

dnewman-polar3d, 2017-07-04:
Actually, you may not be connecting -- I may have misinterpreted part of the log.  But clearly, there's an issue with the crypto libraries and that's your underlying problem.

nyco007, 2017-07-04:
I've followed your instructions which you have summarized:
Download and flash to the BBB bone-debian-8.7-lxqt-4gb-armhf-2017-03-19-4gb

Note that the default account/password is debian/temppwd.

Before installing Octoprint and the Polar Cloud plugin, do

 sudo apt-get update
 sudo apt-get install build-essential python-dev python-setuptools python-pip \
        python-smbus gstreamer1.0-tools libx264-dev gstreamer1.0-plugins-good \
        gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly \
        zlib1g-dev libjpeg-dev python-libtiff libfreetype6-dev \
        liblcms2-dev libwebp-dev tcl-dev tk-dev libopenjpeg-dev
 sudo pip install pyOpenSSL
 sudo pip install Pillow
Follow the Octoprint for BBB directions at Setup-BeagleBone-Black-Rev-C-(Jessie)

markwal, 2017-07-04:
I have a theory on this.  It may be related to your default locale.  On my, I believe I'm using a en_US.UTF-8 and I think that's the automatic encoding for the the challenge.  I think we shouldn't be leaving it to the users settings though since we really just want to use it as a string of bytes.  Unfortunately, SocketIO is giving it back to use as unicode regardless of its form over the wire.  So I *think* what I want to do is translate it to str via unicode.encode('utf-8').  I'm trying that out now and I'll push it to master soon.

@nyco007  what do you think?  Plausible that you're using a non utf-8 default locale?

nyco007, 2017-07-04:
I recognized that when doing sudo pip install pyOpenSSL, the system says:
debian@beaglebone:~$ sudo pip install pyOpenSSL
Requirement already satisfied: pyOpenSSL in /usr/lib/python2.7/dist-packages
Requirement already satisfied: cryptography>=0.2.1 in /usr/lib/python2.7/dist-packages (from pyOpenSSL)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python2.7/dist-packages (from pyOpenSSL)
Requirement already satisfied: cffi>=0.8 in /usr/lib/python2.7/dist-packages (from cryptography>=0.2.1->pyOpenSSL)
Requirement already satisfied: pycparser in /usr/lib/python2.7/dist-packages (from cffi>=0.8->cryptography>=0.2.1->pyOpenSSL)

Is that okay/wanted?

nyco007, 2017-07-04:
@markwal I can't tell you because I don't even know what that is ;D How can I find out or even change that?

markwal, 2017-07-04:
Maybe you could update the plugin to 0.2.5 and try again?

To do that, go to OctoPrint->Settings->Software Update->Check for Updates now.  If it isn't picking up 0.2.5, click on "Advanced" at the bottom of Software Update and choose "Force Check..."

markwal, 2017-07-04:
Hang on.  0.2.5. doesn't have the fix.  :-(

nyco007, 2017-07-05:
I gave @dnewman-polar3d summarized instructions a new try...(format SD card, resize it after flashing and then do following instructions)...somehow it works now. Many thank for your help also to @markwal !

> So, to summarize for a Beagle Bone Black
>
> Download and flash to the BBB bone-debian-8.7-lxqt-4gb-armhf-2017-03-19-4gb
>
> Note that the default account/password is debian/temppwd.
>
> Before installing [...]

markwal, 2017-07-05:
OK.  Thanks.

closed 2017-07-05 by markwal