markwal/OctoPrint-PolarCloud #14: Failed to register OctoPrint with the Polar Cloud.

terracedr, 2017-06-28:
not sure why but this keeps occurring when I try to register my polar account. 

Failed to register OctoPrint with the Polar Cloud.

markwal, 2017-06-28:
What does it say in your logs?

vmorrisPolar, 2017-07-02:
Mark - having this issue as well, my logs say:

2017-07-02 04:09:46,058 - octoprint.plugins.polarcloud - INFO - emit register
2017-07-02 04:10:33,925 - octoprint.plugins.polarcloud - INFO - emit register
2017-07-02 04:10:34,780 - octoprint.plugins.polarcloud - INFO - emit register
2017-07-02 04:11:22,941 - octoprint.plugins.polarcloud - INFO - emit register
2017-07-02 04:11:23,733 - octoprint.plugins.polarcloud - INFO - emit register
2017-07-02 04:11:39,053 - octoprint.plugins.polarcloud - INFO - emit register
2017-07-02 04:12:39,266 - octoprint.plugins.polarcloud - INFO - emit register
2017-07-02 04:14:01,508 - octoprint.plugins.polarcloud - INFO - emit register

markwal, 2017-07-02:
I just pushed a new version that gives better error messages. If you uninstall and reinstall the plugin, you'll get the new version.  If the error message isn't helpful, turn on verbose logging in the plugin settings and then let's see the logs again.

Soon I'll bump the version number and then upgrading will be easier, but I want to get timelapses working before I do that.

markwal, 2017-07-02:
This issue may be resolved (at least vmorrisPolar's version is), but several fixes have been pushed.  @terracedr let me know if the steps I suggested don't resolve it for you.

closed 2017-07-02 by markwal