markwal/OctoPrint-PolarCloud #16: Installed but not visible

ManuelW77, 2017-07-03:
Hey Mark,

i installed the Plugin successfully, but i can't see it in Plugins then.
Also it comes up in Install Plugins as uninstalled.

markwal, 2017-07-03:
I'd like to see octoprint.log.  You can upload it somewhere like Dropbox, Google Drive, OneDrive, share it via link and post a link here.  You could also copy and paste from it to gist.github.com or hastebin.com and put a link here.  If you go copy and paste route make sure you grab everything from the last "Starting OctoPrint" through the end.

ManuelW77, 2017-07-04:
Hey Mark,

so i found, that crypto isn't installed in python. I'll install and try again :)

> 2017-07-04 12:10:36,079 - octoprint.plugin.core - ERROR - Error loading plugin polarcloud

> Traceback (most recent call last):
>   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugin/core.py", line 689, in _import_plugin
>     instance = imp.load_module(key, f, filename, description)
>   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 42, in <module>
>     from OpenSSL import crypto
>   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OpenSSL/__init__.py", line 8, in <module>
>     from OpenSSL import rand, crypto, SSL
>   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OpenSSL/rand.py", line 12, in <module>
>     from OpenSSL._util import (
>   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OpenSSL/_util.py", line 6, in <module>
>     from cryptography.hazmat.bindings.openssl.binding import Binding
> ImportError: No module named cryptography.hazmat.bindings.openssl.binding

ManuelW77, 2017-07-04:
I installed crypte and now it works well. 
Sorry for trouble :)

closed 2017-07-04 by ManuelW77