markwal/OctoPrint-PolarCloud #18: Required Octoslack field not populated

terracedr, 2017-07-08:
how do I get past this?

dnewman-polar3d, 2017-07-08:
Do you have the Octoslack plugin installed?  On the surface, it sounds like an issue with that plugin.  There's certainly nothing in the Polar-Cloud plugin which would generate an "Octoslack" message.

terracedr, 2017-07-08:
yes i do but i do not understand what it is. this message occurs after i register with polar cloud.

terracedr, 2017-07-08:
When using the API Token for Slack connectivity, a Slack channel must be provided

terracedr, 2017-07-08:
should i just delete this plug in?

dnewman-polar3d, 2017-07-08:
I'm waiting for Mark to answer.  Honestly, this sounds like some oddness with the slack plugin.  The Polar Cloud plugin in no way, shape, or form uses Slack or Mattermost.   So, it's hard to imagine how the Polar Cloud plugin would be interfering.  This may be a coincidence that the Octoslack plugin just happened to exhibit an issue with your Slack API Token at the same time.   Or, it may be that some Python module was updated and that has exacerbated a bug in the Octoslack plugin.  For example, it's quite possible that the pyOpenSSL plugin got updated.

terracedr, 2017-07-08:
okay . So I just deleted the slack plugin and re saved my Polar Cloud registration and it did not pop up with that message again and went right to Octoprint startup . So I will proceed with the rest of the install by  enabling Polar Cloud timelapses... Thanks

markwal, 2017-07-10:
I suspect an issue with the Octoslack plugin, rather than with the PolarCloud one since we don't try to connect with Slack at all.

closed 2017-07-10 by markwal