markwal/OctoPrint-PolarCloud #19: Could not parse output from pip

Faramir82, 2017-07-09:
I have OctoPrint installed on a NTC C.H.I.P., it runs great, but I'm having some issues installing this. I installed it once before, uninstalled it but now I'm unable to re-install it.

I have tried to remove the python-cffi and then re-install with no help.

> Installing plugin "PolarCloud (Beta)" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
/home/chip/OctoPrint/venv/bin/python -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Requirement already satisfied: OctoPrint in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg (from OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: SocketIO-client in /home/chip/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint-PolarCloud==0.2.7Beta)
Collecting pyopenssl (from OctoPrint-PolarCloud==0.2.7Beta)
Using cached pyOpenSSL-17.1.0-py2.py3-none-any.whl
Collecting Pillow (from OctoPrint-PolarCloud==0.2.7Beta)
Using cached Pillow-4.2.1.tar.gz
Requirement already satisfied: flask<0.11,>=0.9 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: tornado==4.0.2 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: sockjs-tornado<1.1,>=1.0.2 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: pyserial<2.8,>=2.7 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: netifaces<0.11,>=0.10 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: rsa<3.3,>=3.2 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: requests<2.8,>=2.7 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: psutil<3.3,>=3.2.1 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/psutil-3.2.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: Click<6.3,>=6.2 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: future<0.16,>=0.15 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: websocket-client<0.41,>=0.40 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: python-dateutil<2.7,>=2.6 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/python_dateutil-2.6.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: six in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/six-1.10.0-py2.7.egg (from SocketIO-client->OctoPrint-PolarCloud==0.2.7Beta)
Collecting cryptography>=1.9 (from pyopenssl->OctoPrint-PolarCloud==0.2.7Beta)
Using cached cryptography-1.9.tar.gz
Collecting olefile (from Pillow->OctoPrint-PolarCloud==0.2.7Beta)
Using cached olefile-0.44.zip
Requirement already satisfied: itsdangerous>=0.21 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/itsdangerous-0.24-py2.7.egg (from flask<0.11,>=0.9->OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: MarkupSafe in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/MarkupSafe-1.0-py2.7-linux-armv7l.egg (from Jinja2<2.9,>=2.8->OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: certifi in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/certifi-2017.04.17-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: backports.ssl_match_hostname in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/backports.ssl_match_hostname-3.5.0.1-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: blinker in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/blinker-1.4-py2.7.egg (from Flask-Principal<0.4,>=0.3.5->OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
Requirement already satisfied: Babel>=1.0 in /home/chip/OctoPrint/venv/lib/python2.7/site-packages/Babel-2.4.0-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-PolarCloud==0.2.7Beta)
[... cut ...]
>Command "/home/chip/OctoPrint/venv/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-n_uG1n/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-ERtTRs-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/chip/OctoPrint/venv/include/site/python2.7/cffi" failed with error code 1 in /tmp/pip-build-n_uG1n/cffi/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

WHAAAT, 2017-07-09:
same problem here with raspberry pi 2

markwal, 2017-07-10:
pip is trying to compile the cffi module from source and is unable to find one of the dependencies.  This is less surprising for the CHIP than for the Raspberry Pi since most folks running on a Raspberry Pi are running from the OctoPi image and those libraries are either already there or not needed since the requisite module is already available.

On the command line try:

sudo apt-get install libffi-dev

And then:

source /home/chip/OctoPrint/venv/bin/activate
pip install cffi

Note for @WHAAAT: the path up there to the venv is most likely different for you.  Are you running on an OctoPi image?  If so, do you know what version?

If that gives the same error, you may want to figure out where on your system libffi-dev dropped the ffi.h.  I think it is supposed to land here: /usr/include/libffi/include in which case, this might do the trick:

CFLAGS=-I/usr/include/libffi/include pip install cffi

Unfortunately that may give you a newer version of cffi than pip is trying to install as a dependency of the version of pyOpenSSL that you already have.  And you may either want to go backwards and remove the cffi and try to figure out the version to install or move forward and try to bring pyOpenSSL up to date.  Let me know where you're at after the above.

WHAAAT, 2017-07-10:
Hi,
I´m running from an octopi image. putty tells me:
Linux octopi with kernel 3.18.11-v7+ 
the current octoprint version is: OctoPrint: 1.3.4

Faramir82, 2017-07-10:
sudo apt-get install libffi-dev

It was already installed

source /home/chip/OctoPrint/venv/bin/activate
pip install cffi

sudo apt-get install build-essential libssl-dev libffi-dev python-dev
I needed these updated to install cryptography

pip install cryptography

sudo apt-get install libtiff5-dev libjpeg62-turbo-dev zlib1g-dev libfreetype6-dev liblcms2-dev libwebp-dev libharfbuzz-dev libfribidi-dev tcl8.6-dev tk8.6-dev python-tk
I needed some of these to install Pillow

pip install Pillow

Attempted to install PolarCloud again and was successful!! I'm restarting the octoprint service right now.
Thank You for the help and leading me in the right direction.

win7viena, 2017-07-15:
@markwal Same problem on Orange Pi Zero.
I tried this
_sudo apt-get install libffi-dev_

And then:
_source /home/chip/OctoPrint/venv/bin/activate
pip install cffi_

and looks like it works. Thanks

vmorrisPolar, 2020-12-17:
Should have been fixed in more recent releases.

closed 2020-12-17 by vmorrisPolar