markwal/OctoPrint-PolarCloud #22: Can't Install the Plug-in. Help

TheNFMan, 2017-10-18:
~/oprint/bin/pip install  "https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip"
Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
  Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip (41kB)
    100% |████████████████████████████████| 51kB 463kB/s
Requirement already satisfied: OctoPrint in ./oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg (from OctoPrint-PolarCloud==1.3)
Collecting SocketIO-client (from OctoPrint-PolarCloud==1.3)
  Using cached socketIO-client-0.7.2.tar.gz
Collecting pyopenssl (from OctoPrint-PolarCloud==1.3)
  Using cached pyOpenSSL-17.3.0-py2.py3-none-any.whl
Requirement already satisfied: Pillow in ./oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.3)
Requirement already satisfied: flask<0.11,>=0.9 in ./oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Jinja2<2.9,>=2.8 in ./oprint/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in ./oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: tornado==4.0.2 in ./oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: sockjs-tornado<1.1,>=1.0.2 in ./oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: PyYAML<3.11,>=3.10 in ./oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in ./oprint/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in ./oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in ./oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in ./oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: markdown<2.7,>=2.6.4 in ./oprint/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pyserial<2.8,>=2.7 in ./oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: netaddr<0.8,>=0.7.17 in ./oprint/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in ./oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: sarge<0.2,>=0.1.4 in ./oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: netifaces<0.11,>=0.10 in ./oprint/lib/python2.7/site-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pylru<1.1,>=1.0.9 in ./oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: rsa<3.3,>=3.2 in ./oprint/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pkginfo<1.3,>=1.2.1 in ./oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: requests<2.8,>=2.7 in ./oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in ./oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: psutil<3.3,>=3.2.1 in ./oprint/lib/python2.7/site-packages/psutil-3.2.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Click<6.3,>=6.2 in ./oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in ./oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in ./oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in ./oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: future<0.16,>=0.15 in ./oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: scandir<1.4,>=1.3 in ./oprint/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: websocket-client<0.41,>=0.40 in ./oprint/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: python-dateutil<2.7,>=2.6 in ./oprint/lib/python2.7/site-packages/python_dateutil-2.6.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: six in ./oprint/lib/python2.7/site-packages/six-1.10.0-py2.7.egg (from SocketIO-client->OctoPrint-PolarCloud==1.3)
Collecting cryptography>=1.9 (from pyopenssl->OctoPrint-PolarCloud==1.3)
  Using cached cryptography-2.1.1.tar.gz
    Complete output from command python setup.py egg_info:
    error in cryptography setup command: Invalid environment marker: python_version < '3'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-hQsCaX/cryptography/

Quas7, 2017-10-20:
Same here with Octoprint Version 1.3.5 (master branch)

TheNFMan, 2017-10-20:
I fixed it

TheNFMan, 2017-10-20:
The solution is here https://github.com/markwal/OctoPrint-PolarCloud/issues/19

TheNFMan, 2017-10-20:
Do the install listed with sudo

Quas7, 2017-10-21:
I just tried on raspberry pi 3:
sudo pip install cryptography

closed 2017-11-01 by dnewman-polar3d

kaju666, 2017-12-08:
This helped me:

```
sudo apt-get install libffi-dev

source /home/pi/oprint/bin/activate

pip install cffi

sudo apt-get install build-essential libssl-dev libffi-dev python-dev

pip install --upgrade setuptools

pip install cryptography
```