markwal/OctoPrint-PolarCloud #24: Installing Polar Cloud plugin on Makergear M3 running Octoprint

pmckinney4, 2017-10-27:
Hello,

I am trying to install the PolarCloud plugin on a Makergear M3.  I am not good with Linux.  I know enough to ssh in and run commands, but do not fully understand the commands.   I tried running:

sudo pip install --upgrade cffi
sudo pip install --upgrade pyOpenSSL

When running the --upgrade pyOpenSSL I get:

Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-umjk_K/cryptography
Storing debug log for failure in /root/.pip/pip.log

When trying to install the plugin through the plugin manager in Octoprint I get:

Installing plugin...
Installing plugin "Polar Cloud" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
Installation needs to process external dependencies, that might make it take a bit longer than usual depending on the pip version
/home/pi/oprint/bin/python -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip (41kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg (from OctoPrint-PolarCloud==1.3)
Collecting SocketIO-client (from OctoPrint-PolarCloud==1.3)
Using cached socketIO-client-0.7.2.tar.gz
Requirement already satisfied: pyopenssl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.3)
Exception:
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
status = self.run(options, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 324, in run
requirement_set.prepare_files(finder)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
ignore_dependencies=self.ignore_dependencies))
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 666, in _prepare_file
check_dist_requires_python(dist)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/packaging.py", line 48, in check_dist_requires_python
feed_parser.feed(metadata)
File "/usr/lib/python2.7/email/feedparser.py", line 177, in feed
self._input.push(data)
File "/usr/lib/python2.7/email/feedparser.py", line 99, in push
parts = data.splitlines(True)
AttributeError: 'NoneType' object has no attribute 'splitlines'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Any advice would be appreciated.  I am trying to determine if PolarCloud would be a good program to help manage my 3d printers in a lab setting.

dnewman-polar3d, 2017-10-27:
FWIW, the problem you saw when you tried the manual install of pyOpenSSL is a known interaction problem between the OctoPi image, some versions of pip, and pyOpenSSL as covered in issue #1 .  This plugin was modified to work around that specific issue when it installs itself.  I.e., it avoids the issue you saw with doing a manual pyOpenSSL install.

The issue you are seeing here may simply relate to your having a version of pip which is not compatible with the version of OctoPrint.  That is, I believe you have a newer version of pip installed and OctoPrint cannot parse some of its output causing the install process to then fail.

Hopefully markwal will chime in here shortly as he is far more familiar with OctoPrint installs than I am.

dnewman-polar3d, 2017-10-31:
Mark has confirmed that this is version mismatches between current/available versions of pyOpenSSL, pip, and OctoPrint.  He's built new wheels for Pi 2 and Pi 3.  He'll do Pi Zero when he next has some free time.

In the meantime, you might try installing the plugin again from OctoPrint.

markwal, 2017-11-01:
I've also put up the wheels for Pi Zero now too.  @pmckinney4 have you had a chance to try it again?

markwal, 2018-06-16:
I believe this should be working at the current time.  I've tested with Octopi 0.15

You may need to have the following checkbox, checked:

Under PluginManager->Get More...->Advanced options:
Use --process-dependency-links with pip install

closed 2018-06-16 by markwal