markwal/OctoPrint-PolarCloud #27: install problem

pappysven, 2017-11-10:
when i try to install with the graphic interface:
Installing plugin "Polar Cloud" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
Installation needs to process external dependencies, that might make it take a bit longer than usual depending on the pip version
/home/pi/oprint/bin/python -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip (41kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg (from OctoPrint-PolarCloud==1.3)
Collecting SocketIO-client (from OctoPrint-PolarCloud==1.3)
Using cached socketIO-client-0.7.2.tar.gz
Collecting pyopenssl (from OctoPrint-PolarCloud==1.3)
Using cached pyOpenSSL-17.3.0-py2.py3-none-any.whl
Collecting Pillow (from OctoPrint-PolarCloud==1.3)
Using cached Pillow-4.3.0.tar.gz
Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: sockjs-tornado<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: requests<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: psutil<3.3,>=3.2.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-3.2.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: websocket-client<0.41,>=0.40 in /home/pi/oprint/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: python-dateutil<2.7,>=2.6 in /home/pi/oprint/lib/python2.7/site-packages/python_dateutil-2.6.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: wrapt<1.11,>=1.10.10 in /home/pi/oprint/lib/python2.7/site-packages/wrapt-1.10.11-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: futures<3.2,>=3.1.1 in /home/pi/oprint/lib/python2.7/site-packages/futures-3.1.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages/six-1.10.0-py2.7.egg (from SocketIO-client->OctoPrint-PolarCloud==1.3)
Collecting cryptography>=1.9 (from pyopenssl->OctoPrint-PolarCloud==1.3)
Using cached cryptography-2.1.3.tar.gz
Complete output from command python setup.py egg_info:
error in cryptography setup command: Invalid environment marker: python_version < '3'
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-el3EIF/cryptography/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

i'm a total noob, what must i do please ?

skellep193, 2017-11-15:
Experiencing the same issue

ErikDRuud, 2017-11-20:
I just ran into the same error.
I am assuming that the version of Python on the OctoPi image needs to be update to at least version 3, but I did not have time this morning to check into that.

alex-gibson, 2017-11-21:
Same issue here.  Freshly installed Octopi 0.14 on a Raspberry Pi 2, identical error.

The only other plugin I have installed is Octoprint Anywhere, which loaded fine - but I got a PIP error trying to update this on my previous Octopi 0.13 install.  Seems like PIP is an  issue at the moment...

SteamLearningLabs, 2017-11-22:
Same problem.

markwal, 2017-11-26:
I believe it is due to a new version of cryptography being released and it is incompatible with the setuptools that comes on the Octopi.

I can't put a requirement for setuptools in my setup.py since it is already loaded by the time it reads that (and it is setuptools reading the requirements list anyway).  I should pin the cryptography module so that I have to take manual steps when they upgrade.  I'll put that in the next release.

For now though, this should be fixed for people with raspberry pi's since I put a new built version of cryptography on the referenced wheelhouse.

markwal, 2017-11-29:
It turns out that in addition to the new version of cryptography, the combination of setuptools and pip that are found in octopi 0.14 reveals a bug in the OctoPrint plugin manager.  @foosel has already quickly diagnosed and fixed it (https://github.com/foosel/OctoPrint/commit/14bd0632c041a3e11f2c2056e70b88fd7ea184b8) and it will be in 1.3.6.

Until that fix arrives, we need a workaround.  I'd like to find a way to have my setup.py cause an upgrade of setuptools automatically, but so far my experiments with "setup_requires" don't appear to work.  I think unfortunately, the old version of setuptools is already loaded by that point and doesn't reload.  You can use the command line to upgrade setuptools until I figure something out:

ssh into your raspberry pi, then use these commands:
source ~/oprint/bin/activate
pip install -u setuptools

If that completes successfully, then try installing the plugin from the plugin manager again.

ErikDRuud, 2017-11-29:
That appears to have worked accept that it should be an upper case U in the pip command.

pip install -U setuptools

markwal, 2017-11-29:
Thanks @ErikDRuud.  Updated the workaround comment above to reflect -U.

ErikDRuud, 2017-11-29:
Thank you for the workaround.

SteamLearningLabs, 2017-11-29:
This worked!

chan2000justin, 2017-12-06:
This caused by the old version of "setuptools" inside the "virtualenv". 
Running this may help (at least work for me):
`cd ~/oprint/bin`
`sudo ./pip install --upgrade setuptools`

Hope it's helpful!
Thanks

dnewman-polar3d, 2017-12-13:
This seems to be fixed on the 0.14 Octopi image with OctoPrint 1.3.6 (official release).  However, there's an entirely new issue (#31).

closed 2017-12-13 by dnewman-polar3d