markwal/OctoPrint-PolarCloud #28: Support does not appear to be correctly enabled when slicing

dnewman-polar3d, 2017-11-22:
I have a print which in the Polar Cloud has support enabled

    nozzleSize=400
    layerThickness=300
    printSpeed=50
    supportType=1
    infillSpeed=40
    infillOverlap=15
    filamentDiameter=1750
    filamentFlow=100
    retractionAmountExtruderSwitch=2000
    retractionAmount=1500
    retractionSpeed=30
    initialLayerThickness=350
    extrusionWidth=400
    insetCount=2
    downSkinCount=4
    upSkinCount=4
    initialSpeedupLayers=4
    initialLayerSpeed=15
    inset0Speed=30
    insetXSpeed=35
    moveSpeed=120
    minimalLayerTime=20
    infillPattern=2
    layer0extrusionWidth=600
    spiralizeMode=0
    supportEverywhere=0
    sparseInfillLineDistance=400
    multiVolumeOverlap=0
    enableOozeShield=0
    gcodeFlavor=0
    autoCenter=0
    fixHorrible=1
    objectSink=0
    posx=0
    posy=0
    extruderOffset[0].X=0
    extruderOffset[0].Y=0
    retractionMinimalDistance=1500
    retractionZHop=100
    minimalExtrusionBeforeRetraction=20
    enableCombing=1
    fanFullOnLayerNr=4
    minimalFeedrate=10
    coolHeadLift=0
    fanSpeedMin=100
    fanSpeedMax=100
    skirtDistance=3000
    skirtMinLength=150000
    skirtLineCount=1
    supportAngle=60
    supportXYDistance=800
    supportZDistance=300
    supportLineDistance=4000
    startCode = """
    M73 P0
    (@open_start_gcode)
    M107
    G130 X118 Y118 A118 (Set Stepper motor Vref to defaults)
    G162 X Y F2000 (home XY axes maximum)
    G161 Z F900 (home Z axis minimum)
    G92 X0 Y0 Z-5 A0 (set Z to -5)
    G1 Z0.0 F900(move Z to '0')
    G161 Z F100(home Z axis minimum)
    M132 X Y Z (Recall stored home offsets)
    G1 X-105 Y-73 Z30 F3300.0 (move to waiting position)
    G130 X20 Y20 A20 B20 (Lower stepper Vrefs while heating)
    M140 S110 T0
    M134 T0
    M135 T0
    M104 S235 T0
    M133 T0
    G130 X118 Y118 A118 B118 (Set Stepper motor Vref to defaults)
    G1 X-105 Y-73.000 Z0.270 F9000.000 (Extruder Prime Dry Move)
    G1 X105 Y-73 Z0.270 F1800.000 E25.000 (Extruder Prime Start)
    G92 A0 (Reset after prime)
    G1 Z0.300000 F1000
    (@close_start_gcode)
    """
    endCode = """
    M400
    M73 P100
    M140 S0
    M104 S0 T0
    T0
    G28 X Y F2500
    M18
    M70 P5 (We <3 Making Things!)
    M72 P1 ; Play Ta-Da song
    """
    preSwitchExtruderCode = """
    
    """
    postSwitchExtruderCode = """
    
    """

When I manually slice the STL on the OctoPi with`/usr/local/bin/cura_engine` using the `config.ini` from the cloud (shown above), the support is enabled and generated in the resulting gcode. 

However, when the PolarCloud plugin receives the `config.ini` and invokes Cura, the resulting gcode does not have support generated.  Here's the Cura command the plugin generated

    2017-11-22 17:51:30,328 - octoprint.plugins.cura - INFO - Running u"/usr/local/bin/cura_engine -v -p
        -s coolHeadLift=0
        -s downSkinCount=10000
        -s enableCombing=1
        -s endCode=M400\nM73 P100\nM140 S0\nM104 S0 T0\nT0\nG28 X Y F2500\nM18\nM70 P5 (We <3 Making Things!)\nM72 P1 ; Play Ta-Da song\n
        -s extrusionWidth=400.0
        -s fanFullOnLayerNr=2
        -s fanSpeedMax=100
        -s fanSpeedMin=100
        -s filamentDiameter=1750
        -s filamentFlow=100
        -s fixHorrible=1
        -s gcodeFlavor=0
        -s infillOverlap=15
        -s infillSpeed=40
        -s initialLayerSpeed=15
        -s initialLayerThickness=350
        -s initialSpeedupLayers=4
        -s inset0Speed=30
        -s insetCount=2
        -s insetXSpeed=35
        -s layer0extrusionWidth=600
        -s layerThickness=300
        -s minimalExtrusionBeforeRetraction=20
        -s minimalFeedrate=10
        -s minimalLayerTime=20
        -s moveSpeed=120
        -s multiVolumeOverlap=0
        -s objectSink=0
        -s perimeterBeforeInfill=0
        -s postSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added after the T(n)\n
        -s posx=0.0
        -s posy=0.0
        -s preSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added before the T(n)\n
        -s printSpeed=50
        -s retractionAmount=1500
        -s retractionAmountExtruderSwitch=2000
        -s retractionMinimalDistance=1500
        -s retractionSpeed=30
        -s retractionZHop=100
        -s skinSpeed=50
        -s skirtDistance=3000
        -s skirtLineCount=1
        -s skirtMinLength=150000
        -s sparseInfillLineDistance=400.0
        -s startCode=(@ignore 220)\n(@ignore ?print_bed_temperature?)\n\nM73 P0\n(@open_start_gcode)\nM107\nG130 X118 Y118 A118 (Set Stepper motor Vref to defaults)\nG162 X Y F2000 (home XY axes maximum)\nG161 Z F900 (home Z axis minimum)\nG92 X0 Y0 Z-5 A0 (set Z to -5)\nG1 Z0.0 F900(move Z to '0')\nG161 Z F100(home Z axis minimum)\nM132 X Y Z (Recall stored home offsets)\nG1 X-105 Y-73 Z30 F3300.0 (move to waiting position)\nG130 X20 Y20 A20 B20 (Lower stepper Vrefs while heating)\nM140 S110 T0\nM134 T0\nM135 T0\nM104 S235 T0\nM133 T0\nG130 X118 Y118 A118 B118 (Set Stepper motor Vref to defaults)\nG1 X-105 Y-73.000 Z0.270 F9000.000 (Extruder Prime Dry Move)\nG1 X105 Y-73 Z0.270 F1800.000 E25.000 (Extruder Prime Start)\nG92 A0 (Reset after prime)\nG1 Z0.300000 F1000\n(@close_start_gcode)\n
        -s supportAngle=-1
        -s supportEverywhere=0
        -s supportExtruder=-1
        -s supportLineDistance=4000
        -s supportXYDistance=800
        -s supportZDistance=300
        -s upSkinCount=10000
        -o /tmp/tmpb9POlt.gco /home/pi/.octoprint/uploads/polarcloud/current-print.stl" in /usr/local/bin

dnewman-polar3d, 2017-12-07:
Looks like the underlying OctoPrint code, [https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/cura/profile.py#L895](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/cura/profile.py#L895),  expects to see the obsolete "support" setting in order to then set key support parameters.  (In this case "supportAngle").  If you look at Cura 15.04.6 sources, you'll see that it no longer looks for a parameter named "support".   The OctoPrint code wants to see either "support=buildplate" or "support=everywhere" in order to set values like "supportAngle".

To work around this issue in OctoPrint, we need to modify this plugin to look for and passthru the obsolete "support" option.  And further for the Polar Cloud to set it.

dnewman-polar3d, 2017-12-07:
My mistake: here's the bug....

    "supporteverywhere":    ("support",            lambda x: "everywhere" if x else "none") ,

That ignores the case of "support=buildplate".  In that case, supportEverywhere=0 BUT other support values have been supplied.   E.g., supportAngle >= 0.

markwal, 2017-12-07:
Sorry @dnewman-polar3d I've been away on another thing. Thanks for the diagnosis. I'll get this fixed today or tomorrow.

commit 38221932 (2017-12-09): Also generate "touching buildplate" support

Should fix #28

Also fix a typo in rectraction_minimal_extrusion
closed 2017-12-09 by markwal