markwal/OctoPrint-PolarCloud #32: Install problem OctoPi 0.14

Maximvdw, 2017-12-17:
I am having problems installing the plugin on OctoPi 0.14 running OctoPrint 1.3.6

Installation seems to work
`Installing plugin "Polar Cloud" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
Installation needs to process external dependencies, that might make it take a bit longer than usual depending on the pip version
/home/pi/oprint/bin/python -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip --process-dependency-links
Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip (41kB)
DEPRECATION: Dependency Links processing has been deprecated and will be removed in a future release.
Requirement already satisfied (use --upgrade to upgrade): OctoPrint-PolarCloud==1.3 from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip in /home/pi/oprint/lib/python2.7/site-packages
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-PolarCloud==1.3)
Requirement already satisfied: cffi==1.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.3)
Requirement already satisfied: cryptography==2.1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.3)
Requirement already satisfied: SocketIO-client in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pyopenssl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Pillow in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.3)
Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: sockjs-tornado<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: psutil<6,>=5.4.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-5.4.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: websocket-client<0.41,>=0.40 in /home/pi/oprint/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: python-dateutil<2.7,>=2.6 in /home/pi/oprint/lib/python2.7/site-packages/python_dateutil-2.6.0-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: wrapt<1.11,>=1.10.10 in /home/pi/oprint/lib/python2.7/site-packages/wrapt-1.10.11-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: futures<3.2,>=3.1.1 in /home/pi/oprint/lib/python2.7/site-packages/futures-3.1.1-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: emoji<0.5,>=0.4.5 in /home/pi/oprint/lib/python2.7/site-packages/emoji-0.4.5-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: monotonic<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/monotonic-1.3-py2.7.egg (from OctoPrint->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: pycparser in /home/pi/oprint/lib/python2.7/site-packages (from cffi==1.11.2->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: six>=1.4.1 in /home/pi/oprint/lib/python2.7/site-packages/six-1.10.0-py2.7.egg (from cryptography==2.1.4->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: idna>=2.1 in /home/pi/oprint/lib/python2.7/site-packages/idna-2.6-py2.7.egg (from cryptography==2.1.4->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: asn1crypto>=0.21.0 in /home/pi/oprint/lib/python2.7/site-packages (from cryptography==2.1.4->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: ipaddress in /home/pi/oprint/lib/python2.7/site-packages (from cryptography==2.1.4->OctoPrint-PolarCloud==1.3)
Requirement already satisfied: enum34 in /home/pi/oprint/lib/python2.7/site-packages (from cryptography==2.1.4->OctoPrint-PolarCloud==1.3)
[... cut ...]
DEPRECATION: Dependency Links processing has been deprecated and will be removed in a future release.
Installing collected packages: OctoPrint-PolarCloud
Running setup.py install for OctoPrint-PolarCloud: started
Running setup.py install for OctoPrint-PolarCloud: finished with status 'done'
Successfully installed OctoPrint-PolarCloud-1.3
Done!`

Afterwards I restart, but the plugin looks like it is not installed. No settings are available and it isn't listed with the installed plugins.

I can simply just install it again

dnewman-polar3d, 2017-12-17:
1. When did you attempt this install?  I ask as this may have been fixed a couple of days back (Issue #31).
2. After the plugin appears to have installed correctly, does OctoPrint display the message, "Plugin Installed. A plugin was installed successfully, however it was impossible to detect which one. Please Restart OctoPrint to make sure everything will be registered properly."  Point being, does OctoPrint claim it doesn't know which plugin was installed?

If you are able to log in to the Raspberry Pi with ssh, you can

    cd .octoprint/logs
    more octoprint.log

And see if any errors are being reported as the plugin is installed.  (See, for example, second comment posted for Issue #31.)

dnewman-polar3d, 2017-12-18:
2017-12-18 00:39:55,930 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
    2017-12-18 00:42:26,075 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugin/core.py:1538:     UserWarning: No PKG-INFO found for package: octoprint_polarcloud
      warnings.warn('No PKG-INFO found for package: %s' % self.package_name)

    2017-12-18 00:42:26,097 - octoprint.plugin.core - ERROR - Error loading plugin polarcloud
    Traceback (most recent call last):
      File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugin/core.py", line 732, in _import_plugin
        instance = imp.load_module(key, f, filename, description)
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 49, in <module>
        from PIL import Image
      File "/home/pi/oprint/local/lib/python2.7/site-packages/PIL/Image.py", line 56, in <module>
        from . import _imaging as core
    ImportError: libjpeg.so.8: cannot open shared object file: No such file or directory
    2017-12-18 00:42:26,112 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. 
    Please restart OctoPrint.

dnewman-polar3d, 2017-12-22:
With the latest (pillow?) wheel, this problem appears to be resolved.  Thanks!

closed 2017-12-22 by dnewman-polar3d