markwal/OctoPrint-PolarCloud #34: Camera view issues

naviathan, 2018-01-03:
When selecting Printers the icon view of the camera shot does not reflect the camera flip settings. Once selected the view is properly rotated (this requires gstreamer for some reason), but the viewer is in the center of the screen.
[image]
[image]

dnewman-polar3d, 2018-01-03:
Thanks for the report!  I've pointed our web frontend engineer at this and hopefully we'll correct this shortly (as in next week; he's out sick at the moment).

naviathan, 2018-01-03:
Awesome, thanks for the quick response!

dnewman-polar3d, 2018-01-06:
When you have a chance, see if this issue is now resolved at [https://polar3d.com](https://polar3d.com).

naviathan, 2018-01-06:
Perfect! Thank you for the quick response. Can't even get that kind of service from products I pay hundreds of dollars for.

dnewman-polar3d, 2018-01-06:
Thanks for the followup and kind words.  It is appreciated.  I'll go ahead and close this issue out.  Feel free to re-open should you find it's not working after all.

closed 2018-01-06 by dnewman-polar3d