markwal/OctoPrint-PolarCloud #35: sendNextPrint command be added

Will695, 2018-01-28:
Please can the sendNextPrint command be added and implemented to the the OctoPrint-PolarCloud Plugin. Basically so I can load up the print queue with gcode and it will go through them without stopping for human interaction

dnewman-polar3d, 2018-01-29:
1. It's okay to use this with the Polar Cloud plugin.  No need to require the capability in capabilities.  (Mark, I can explain why in private e-mail.)
2. Would likely just want an option whereby an OctoPrint user can enable this in the plugin.  With suitable warning that they need to have some means of automatically clearly their build plate before the print actually finishes.  Because, once enabled, a new print will be started each time a print finishes.

Will695, 2018-02-07:
Yes that would be awesome if this could be implemented! Would save me loads of time printing!

Will695, 2018-02-16:
any progress on this as I would love to use it.?

Will695, 2018-05-13:
Hi, 
just checking to see if there's any progress with this? 

Will

vmorrisPolar, 2020-12-17:
@alexp3d @wjsteele would this also satisfy the Creality PrintMill belt work?

vmorrisPolar, 2021-01-28:
@Will695 we're adding this, it was already defined in the plugin, we'll be giving users an option to tell us whether or not their printer IS a belt/continuous printer, and if you want the continuous printing on or off.