markwal/OctoPrint-PolarCloud #38: PolarCloud plugin not showing up after successful install

aknight55, 2018-04-16:
OctoPrint v. 1.3.8,  OctoPi 0.15.0

The PolarCloud plugin installs successfully but does not show up under Plugins.  From viewing the log it says the Plugin installed successfully and I've reinstalled it a couple of times but it never shows up under Plugins.
2018-04-16 19:25:37,236 * Installation needs to process external dependencies, that might make it take a bit longer than usual depending on the pip version
2018-04-16 19:25:37,239   /home/pi/oprint/bin/python2 -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip --process-dependency-links
2018-04-16 19:25:42,627 > Collecting https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
2018-04-16 19:25:42,628 > Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip
2018-04-16 19:25:46,363 > Requirement already satisfied (use --upgrade to upgrade): OctoPrint-PolarCloud==1.4 from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip in /home/pi/oprint/lib/python2.7/site-packages
2018-04-16 19:25:46,363 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,363 > Requirement already satisfied: cffi==1.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,364 > Requirement already satisfied: cryptography==2.1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,364 > Requirement already satisfied: SocketIO-client in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,364 > Requirement already satisfied: pyopenssl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,365 > Requirement already satisfied: Pillow in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,365 > Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,365 > Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,365 > Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,366 > Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,366 > Requirement already satisfied: sockjs-tornado<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,366 > Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,366 > Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,367 > Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,367 > Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,367 > Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,367 > Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,368 > Requirement already satisfied: pyserial<3.5,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,368 > Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,368 > Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,369 > Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,369 > Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,369 > Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,369 > Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,370 > Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,370 > Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,370 > Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,370 > Requirement already satisfied: psutil==5.4.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,371 > Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,371 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,371 > Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,371 > Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,372 > Requirement already satisfied: future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,372 > Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,372 > Requirement already satisfied: websocket-client<0.41,>=0.40 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,372 > Requirement already satisfied: python-dateutil<2.7,>=2.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,373 > Requirement already satisfied: wrapt<1.11,>=1.10.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,373 > Requirement already satisfied: futures<3.2,>=3.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,373 > Requirement already satisfied: emoji<0.5,>=0.4.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,373 > Requirement already satisfied: monotonic<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,374 > Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,374 > Requirement already satisfied: pycparser in /home/pi/oprint/lib/python2.7/site-packages (from cffi==1.11.2->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,374 > Requirement already satisfied: six>=1.4.1 in /home/pi/oprint/lib/python2.7/site-packages (from cryptography==2.1.4->OctoPrint-PolarCloud==1.4)
2018-04-16 19:25:46,375 > Requirement already satisfied: ipaddress in /home/pi/oprint/lib/python2.7/site-packages (from cryptography==2.1.4->OctoPrint-PolarCloud==1.4)
[... cut ...]
2018-04-16 19:25:51,677 > Downloading https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip (41kB)
2018-04-16 19:25:53,743 > Installing collected packages: OctoPrint-PolarCloud
2018-04-16 19:25:53,744 > Running setup.py install for OctoPrint-PolarCloud: started
2018-04-16 19:25:54,245 ! DEPRECATION: Dependency Links processing has been deprecated and will be removed in a future release.
2018-04-16 19:25:56,228 > Running setup.py install for OctoPrint-PolarCloud: finished with status 'done'
2018-04-16 19:25:56,228 > Successfully installed OctoPrint-PolarCloud-1.4

4levels, 2018-05-07:
Hi Polar Cloud,

I have the same issue here: plugin installs, OctoPrint reloads (even tried rebooting the pi) but no PolarCloud plugin or settings.
I also tried with the soon to be deprecated OctoPrint feature "Enable legacy plugin asset bundling" but to no avail.

Kind regards,

ubergeekseven, 2018-05-09:
Same here. Installed astroprint after trying for hours to get this working. The astroprint service is not optimal for me, compared to the polar service. 

We use this plugin in our school to combine the polar printers with the rest.

marcushasfun, 2018-05-12:
Same issue here. Tried from plugin manager and command line. No dice.

Plugin manager install throws a dialog that reads:

"Plugin management log
Install Unknown: 
A restart is needed for the changes to take effect."

[image]

wjsteele, 2018-05-21:
No. Mark Walker is the maintainer. I believe he might be OOF on another project. When I return to the states later this week, I will help dive into what is happening. I've seen several successful installs this week, so don't really know what is happening to your particular setups.

Jyckle, 2018-05-22:
I've also had the same issue, but have also had the following show up during installation, which I believe may be part of the problem.  
[image]
The plugin still says it successfully installs, but shows up as Unknown and has the same issue as others have stated

wjsteele, 2018-05-22:
Unfortunately, my flight out of Shanghai has been delayed... but Jyckle's input is very helpful.

There are two options to explore... either PIP needs updating, or you can try to update Cryptography directly.

Connectivity here in Shanghai isn't very good, but here are two places to start.

1) Upgrade PIP: https://stackoverflow.com/questions/41661082/pyopenssl-version-0-13-1-after-pip-upgrade or https://github.com/pypa/pip/issues/3508

2) Upgrade Cryptography: https://cryptography.io/en/latest/faq/

--Bill

Docmani, 2018-05-23:
I am having the same problem as above and I only recently installed Octopi.

I tried to upgrade Cryptography, however when you do that, the Polar Cloud plugin complains that it needs cryptography==2.1.4. So, the workaround might be to downgrade the pyopenssl version so that it works with cryptography 2.1.4. 

Ultimately, the Polar Cloud should be using the newer version, but I don't know if that's a trivial or major change.

wjsteele, 2018-05-23:
That's the thing... I'm pretty sure we didn't change anything... something in the base image changed to cause this to break.

- Bill

Docmani, 2018-05-23:
I would agree, the default version of pyopenssl and matching cryptography 2.2+ that comes with the octopi image appears to be incompatible with the plugin now. I have no doubt the change was on their end.

southernastro, 2018-05-25:
Hi, I have just started with Octoprint and have the same issue

akf-was-here, 2018-05-27:
I got it to show up after doing this (starting after a failed plug-in install):

/home/pi/oprint/bin/python2 -m pip uninstall cryptography
/home/pi/oprint/bin/python2 -m pip install cryptography==2.1.4
/home/pi/oprint/bin/python2 -m pip install cryptography interfaces
/home/pi/oprint/bin/python2 -m pip uninstall cffi
/home/pi/oprint/bin/python2 -m pip install cffi==1.11.2
/home/pi/oprint/bin/python2 -m pip uninstall OctoPrint-PolarCloud

and then installing the PolarCloud package in Octoprint.  Now polar cloud appears in the UI but fails in it's communication to the polar cloud servers.  If I check octoprint.log  I see some new errors associated with this plug-in:

2018-05-27 08:50:30,196 - octoprint.plugins.polarcloud - ERROR - Could not retrieve system commands
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_polarcloud/__init__.py", line 1006, in _custom_command_list
    r = client.get("/api/system/commands")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_client/__init__.py", line 241, in get
    return self.request("GET", path, params=params, timeout=timeout)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_client/__init__.py", line 237, in request
    response = s.send(request, timeout=timeout)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 618, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/adapters.py", line 521, in send
    raise ReadTimeout(e, request=request)
ReadTimeout: HTTPConnectionPool(host='127.0.0.1', port=5000): Read timed out. (read timeout=30)
2018-05-27 08:51:05,357 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages/ (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-05-27 08:51:05,387 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-05-27 08:51:06,577 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2018-05-27 08:51:14,967 - octoprint.plugins.polarcloud - INFO - Socket disconnected, clear and restart
2018-05-27 08:51:14,997 - octoprint.plugins.polarcloud - WARNING - Unable to connect to Polar Cloud

akf-was-here, 2018-05-27:
Seems to be connecting fine now, with no other modifications from above.

jeffstreet, 2018-06-06:
I'm seeing the same problem. 
Didn't attempt @akf-was-here's proceedure due to the final "Unable to connect to Polar Cloud" warning.

akf-was-here, 2018-06-06:
It seems to be working fine to track print progress, review the camera feed and do remote print control.  I don't remotely start or queue print jobs so not sure about that.

wjsteele, 2018-06-06:
Yes, if it's working fine to do those things, then it will do everything, as they're all being done over the same connection.

- Bill

wjsteele, 2018-06-06:
You should go ahead and follow his steps... his was just a temporary network issue, I'm sure... as the cloud hasn't had any issues or downtime.

- Bill

bretcobrageek, 2018-06-12:
Bill, any ETA for a fix for this?  I followed the steps above and have it working nicely, but there are other plugins I'd like to use that won't install due to an out of date version of cryptography.  :-).  Thanks.

markwal, 2018-06-16:
I believe this issue is fixed in the most recent release.

Well, mostly.  The difficulty is that older versions of octopi had broken installs of cryptography and/or an old version of setuptools that isn't supported by the new cryptography modules.  Unfortunately, demanding an upgrade of setuptools isn't something that works from a requirements perspective because setuptools is interpreting the requirements.

So, what I've done is made a custom wheelhouse with binary compiles of the platform dependent modules (cryptography, cffi and Pillow) which I made on a fresh Octopi 0.15 image.  This has two problems:

1. if you've got an older octopi and/or other libraries have been installed, my binary *might* not be compatible which will fail the install.
2. the other issue is that you need to have the "Advanced" option "--process-dependency-links" turned on for the SoftwareUpdater *and* the PluginManager.

If all else fails, the workaround given above is slow, but should work:

```
source ~/oprint/bin/activate
pip uninstall cryptography
pip uninstall cffi
pip uninstall Pillow
pip install cffi=1.11.5
pip install cryptography==2.2.2
pip install Pillow

```

Those last three will take a long time, but then you should be able to update/install this plugin and restart the server and be good to go.

closed 2018-06-16 by markwal

bretcobrageek, 2018-06-18:
Thanks Mark.  Really appreciate the support on this.  But I'm a bit of a noob on this, so I'm not exactly sure what I should do at this point.  I have the latest octopi and then did the comands above listed by "akf-was-here".  So now, should I uninstall the polarcloud plugin and attempt another install?  Or should I do an uninstall of polarcloud plugin, then your commands, then try installing the new plugin?   
Thanks!

markwal, 2018-06-18:
I'd say just try to reinstall the plugin.  You do that from the plugin manager.  Choose the wrench button in the upper right to show plugins that are already installed.  It should show with a reinstall button.  Open the advanced section at the bottom and check the "process dependency links". And then click reinstall.

bretcobrageek, 2018-06-18:
Thanks Mark. Would I need to do something to fix the cryptography version that seems to be causing problems for other plugins I've tried to install?  Or will doing that update of polarcloud just fix it?  :-)

bretcobrageek, 2018-06-18:
I have quite a few plugins installed.  (not in front of my machine currently, so I can't show the list).  I have a couple that display extra information the LCD screen of the printer and at the top of the web page.  I ran into the error when trying to install File-manager.  I have octopi 0.15.1.  I will try updating when I am home in a couple hours and see what happens.  I'll try uninstalling and updating the cryptogrphy version if the normal install doesn't work.

bretcobrageek, 2018-06-19:
Thanks for the great update Mark.  I just did the normal plugin update and it seemed to have  sorted out everything.  Sorry for all the questions when it just worked :-).