markwal/OctoPrint-PolarCloud #40: Dremel 3D45 Polar Clound connection error

jlovett, 2018-07-28:
When trying to connect and verify with Polar Cloud I get the error "Not able to connect, invalid public key supplied. The public key must be a non empty string."  I have put in my email and PIN number and I can't figure out what is wrong.  I even factory reset the printer thinking that it would fix the issue potentially, but the printer retained my login information.

dnewman-polar3d, 2018-07-28:
@jlovett The Dremel printers do not use Octoprint to connect to the Polar Cloud.  You will need to raise this issue with support@polar3d.com or via the [Polar 3D support page](https://polar3d.freshdesk.com/support/home).   I'm closing out this issue here as this Octoprint plugin isn't used by the Dremels.

closed 2018-07-28 by dnewman-polar3d

sunny500, 2018-09-20:
I'm  facing this type of issue too kindly help me out thanks.