markwal/OctoPrint-PolarCloud #43: Plugin doesn't install after saying it installed successfully

skint007, 2018-09-06:
I've been trying to install the plugin multiple and haven't been successful. It says it installed, but I noticed when its finished and asks to restart it says "Installed Unknown" instead of Polar Cloud.

Here is what it looks like:
https://www.dropbox.com/s/qnmrfgczchlfckr/Install_Polar3d.png?dl=0

AbuKareem, 2018-09-06:
I'm having the exact same problem
[image]

AbuKareem, 2018-09-06:
### Update ### 
i followed these commands from issue https://github.com/markwal/OctoPrint-PolarCloud/issues/38
source ~/oprint/bin/activate
pip uninstall cryptography
pip uninstall cffi
pip uninstall Pillow
pip install cffi=1.11.5
pip install cryptography==2.2.2
pip install Pillow

now the install shows the plugin name
but it is still not showing in the plugins menu
and when searching in plugins manager it shows it is not installed and gives the option to install

[image]

AbuKareem, 2018-09-06:
### Update #2 ### 
after the first installation reverted (cryptography) to 2.2.2 
and trying to install the plugin again (because it's not showing) the message after installation displayed the name as unknown 
so perhaps the issue is with the cryptography version that the plugin is using?

stefanuc111, 2018-09-26:
I've the same problem, with octoprint installed in the orange pi lite

markhurlburt, 2018-12-17:
I'm running into this as well!

NovaViper, 2019-02-19:
Same issue, happened on OctoPrint 1.3.0 and 1.3.10

vmorrisPolar, 2020-12-17:
This issue should have been fixed in subsequent releases.

closed 2020-12-17 by vmorrisPolar