markwal/OctoPrint-PolarCloud #47: Installing plugin...

drkbalamurugan, 2019-01-28:
Installing plugin "Polar Cloud" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
Installation needs to process external dependencies, that might make it take a bit longer than usual depending on the pip version
c:\octoprint\venv\scripts\python.exe -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip --no-cache-dir --process-dependency-links
Usage:
c:\octoprint\venv\scripts\python.exe -m pip install [options] <requirement specifier> [package-index-options] ...
c:\octoprint\venv\scripts\python.exe -m pip install [options] -r <requirements file> [package-index-options] ...
c:\octoprint\venv\scripts\python.exe -m pip install [options] [-e] <vcs project url> ...
c:\octoprint\venv\scripts\python.exe -m pip install [options] [-e] <local project path> ...
c:\octoprint\venv\scripts\python.exe -m pip install [options] <archive url/path> ...
no such option: --process-dependency-links
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

CKENT0369, 2019-02-04:
I have the same problem, cannot install polar cloud. 

`

Installing plugin "Polar Cloud" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...Installation needs to process external dependencies, that might make it take a bit longer than usual depending on the pip version/home/pi/oprint/bin/python2 -m pip install https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip --process-dependency-linksUsage:/home/pi/oprint/bin/python2 -m pip install [options] <requirement specifier> [package-index-options] .../home/pi/oprint/bin/python2 -m pip install [options] -r <requirements file> [package-index-options] .../home/pi/oprint/bin/python2 -m pip install [options] [-e] <vcs project url> .../home/pi/oprint/bin/python2 -m pip install [options] [-e] <local project path> .../home/pi/oprint/bin/python2 -m pip install [options] <archive url/path> ...no such option: --process-dependency-linksError!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

`

wimorrison, 2019-02-20:
Just to add my pen’arth I have been completely unable to install this plugin despite trying several time and several ways.

My errors are exactly the same as above - what is the expected resolution time please?

alexp3d, 2019-02-21:
Hi Folks, 
It looks like a recent pip installer deprecation broke a few important dependency installations for the Polar Cloud Plugin. We have identified the issue and are working towards a fix. I hope to have this dealt within the next week or so but will update you as required.

MadderPad, 2019-02-21:
For anyone looking for a work around, and is happy to work with the command line on the Pi.

Running the following commands (I've reversed engineered these from the setup.py for the plugin)
`~/oprint/bin/pip install cffi==1.11.5 --extra-index-url https://markwal.github.io/wheels/`
`~/oprint/bin/pip install cryptography==2.2.2 --extra-index-url https://markwal.github.io/wheels/`
`~/oprint/bin/pip install SocketIO-client --extra-index-url https://markwal.github.io/wheels/`
`~/oprint/bin/pip install pyopenssl --extra-index-url https://markwal.github.io/wheels/`
`~/oprint/bin/pip install Pillow --extra-index-url https://markwal.github.io/wheels/`

Then install "https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip" using the plugin manager in OctoPrint, using the '... From URL' option. (Make sure use --process-dependency-links **isn't** selected in the advanced options ;) )

Once OctoPrint restarted, I was able to set up the PolarCloud plugin as per the documentation.

Paddy

wimorrison, 2019-02-21:
Brilliant!!!

I ran through the command lines and they all went well except

~/oprint/bin/pip install pyopenssl --extra-index-url https://markwal.github.io/wheels/

which threw an error and when I installed Polarcloud the install threw an error about me having 2.2 of something and it wanted 2.3, I think it was related to above line but I couldn't catch it as it scrolled of my screen :(

That said a reboot of the pi and I could set up PolarCloud and it connects.

fingers crossed that the 'proper fix' can install over this at some point?

Iain

MadderPad, 2019-02-21:
I _think_ the openssl messages are warnings rather than errors - but only time will tell.

hlKarln, 2019-02-26:
Thanks for your help MadderPad, I get the same issue and to resolve that I have follow your step.
It work fine ... ;-)

Thanks.

andrerjesus, 2019-03-04:
> I _think_ the openssl messages are warnings rather than errors - but only time will tell.

pi@octopi:~ $ sudo ~/oprint/bin/pip install pyopenssl --extra-index-url https://markwal.github.io/wheels/
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://markwal.github.io/wheels/
Collecting pyopenssl
  Downloading https://files.pythonhosted.org/packages/01/c8/ceb170d81bd3941cbeb9940fc6cc2ef2ca4288d0ca8929ea4db5905d904d/pyOpenSSL-19.0.0-py2.py3-none-any.whl (53kB)
    100% |████████████████████████████████| 61kB 566kB/s
Collecting cryptography>=2.3 (from pyopenssl)
  Downloading https://files.pythonhosted.org/packages/07/ca/bc827c5e55918ad223d59d299fff92f3563476c3b00d0a9157d9c0217449/cryptography-2.6.1.tar.gz (491kB)
    100% |████████████████████████████████| 501kB 299kB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  Complete output from command /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpDnIHUd:
  Traceback (most recent call last):
    File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 207, in <module>
      main()
    File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 197, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 48, in get_requires_for_build_wheel
      backend = _build_backend()
    File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 39, in _build_backend
      obj = getattr(obj, path_part)
  AttributeError: 'module' object has no attribute '__legacy__'

  ----------------------------------------
Command "/home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpDnIHUd" failed with error code 1 in /tmp/pip-install-biixEw/cryptography
pi@octopi:~ $

carlesm, 2019-03-23:
You can skip all errors by going to a pyopenssl that works well with cryptography 2.2, for example the previous version 18.0.0 (19.0.0, the next one is giving you the errors):

`~/oprint/bin/pip install pyopenssl=18.0.0 --extra-index-url https://markwal.github.io/wheels/
`

With that it went without any complaints.

Retoxx, 2019-03-29:
Using MadderPad's list of commands, and substituting the 2nd to the last with a slightly modified Carlesm's  line (it needed the double equal sign) works perfectly!! THANK YOU BOTH!!!!

**~/oprint/bin/pip install cffi==1.11.5 --extra-index-url https://markwal.github.io/wheels/
~/oprint/bin/pip install cryptography==2.2.2 --extra-index-url https://markwal.github.io/wheels/
~/oprint/bin/pip install SocketIO-client --extra-index-url https://markwal.github.io/wheels/
~/oprint/bin/pip install pyopenssl==18.0.0 --extra-index-url https://markwal.github.io/wheels/
~/oprint/bin/pip install Pillow --extra-index-url https://markwal.github.io/wheels/**

carlesm, 2019-03-29:
Yup, my bad (or my clipboard bad, in fact).

wsager42, 2020-08-14:
I have tried to do work around suggested above but I was not successful 
I'm using raspberry pi 4 controller that is connected to an Ender 3 V2 printer
Octoprint 1.4.2 Python 2.7.16 Octopi 0.17.0 
If you guys have any suggestions I would greatly appreciate it, I have been trying to get polar cloud to install all week with no success!!

closed 2021-01-28 by vmorrisPolar