markwal/OctoPrint-PolarCloud #48: Failed to register OctoPrint with the Polar Cloud

mratingittar, 2019-08-18:
Whenever I try to connect my printer to Polar Cloud I get this error message:

"Failed to register OctoPrint with the Polar Cloud. The e-mail address and/or the PIN are not recognized by Polar Cloud."

This is my log:

2019-08-18 18:51:48,341 - octoprint.plugins.polarcloud - DEBUG - key_filename: /home/pi/.octoprint/data/polarcloud/p3d_key
2019-08-18 18:51:48,346 - octoprint.plugins.polarcloud - INFO - emit register
2019-08-18 18:51:49,497 - octoprint.plugins.polarcloud - DEBUG - on_register_response: {u'status': u'FAILED', u'reason': u'EMAIL_PIN_ERROR'}
2019-08-18 18:52:01,945 - octoprint.plugins.polarcloud - DEBUG - self._socket: <socketIO_client.SocketIO object at 0x6c399310>
2019-08-18 18:52:01,948 - octoprint.plugins.polarcloud - DEBUG - _wait_and_process

I have tried uninstalling and reinstalling the plugin once, and tried another e-mail address on Polar Cloud. Anything else I can try?

JaredWith1r, 2020-05-08:
I am having this same issue, looking for any updates.

markwal, 2020-05-10:
Hmmmm....  Is it a printer that has ever been connected to the Polar Cloud?  Does it already show in the Polar UI?  One possibility is that OctoPrint has stale data stored about a previous mismatching registration.  I could walk you through cleaning out the plugin data, but perhaps I should add a button to do that.

JaredWith1r, 2020-05-10:
It has never been connected to Polar Cloud, and it was not in the UI. 

However, I went to Plugin Manager and cleared the plugin data. Went back to my Polar Cloud account and noticed my PIN had not changed when I changed it manually. So I used the pin it said, and it works now! 

Not sure if that is an issue or what, but at least I am in now.

closed 2020-05-10 by markwal