markwal/OctoPrint-PolarCloud #49: Make gcode files from Octoprint available for print from PolarCloud

yordanov-alexandar, 2020-04-03:
I would like to be able to reprint some of my existing Gcode files that reside on the Octoprint machine while on the go

markwal, 2020-05-10:
Well.... you sorta can, you can download them from the OctoPrint UI and then upload them into PolarCloud via the "G-Code Job" button.  When you click on a printer in the Polar UI, there is a "G-Code Job" button in the upper left hand corner.

But I'm guessing, what you'd like is for Polar to show you a list of available gcode from the OctoPrint box and let you start one of those.  We'd need to ask the Polar guys for that feature though.

vmorrisPolar, 2020-12-17:
Hi @yordanov-alexandar, this is Van from Polar. We'll think about this and do some digging to see what work would be involved, we'll report back here if we work it into the development schedule.

closed 2020-12-17 by vmorrisPolar