markwal/OctoPrint-PolarCloud #50: Plugin not installing in plugin directory same as issue #43

Onehunterguy, 2020-08-08:
Have installed plug in 6 times with the same results as issue #43 from 2018. Issue with recent Octoprint update 1.4.2?

[image]
[image]

wjsteele, 2020-08-08:
Have you followed the resolution steps as documented in #43?

Onehunterguy, 2020-08-08:
I haven’t, only because it was two years ago and I assume the work around would have already been Implemented.
Was hoping for an update....
It’s a current instance of Octoprint on a Pi 4

Dave

wjsteele, 2020-08-08:
The work around was documented in those steps in #38.  The issue isn't with the Polar Cloud plug-in, but it's an issue with one of the base system's dependencies, which we are not in control of. 

Bill

Onehunterguy, 2020-08-08:
Thanks Bill

Dave

reuter, 2020-08-11:
I'm having the same issue as described here. I tried following the steps on 38 twice now and it's still not working.
Anything else I can try?

wjsteele, 2020-08-11:
You'll need to post more info... without logs there is no way to know what the actual issue is, as the issue described in #38 is very specific to that one crypto library being out of date.  If that's not working, you'll want to open a new issue with the details in it.

reuter, 2020-08-11:
Fair enough, here's the log. I'm happy to open a new one if I need to.

> 2020-08-11 10:17:06,479   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpWcKN_t/OctoPrint-PolarCloud-master.zip --no-cache-dir
> 2020-08-11 10:17:11,008 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> 2020-08-11 10:17:11,009 > Processing /tmp/tmpWcKN_t/OctoPrint-PolarCloud-master.zip
> 2020-08-11 10:17:13,336 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (1.4.2)
> 2020-08-11 10:17:17,307 > Requirement already satisfied: cryptography@ https://markwal.github.io/wheelhouse/cryptography-2.5-cp27-none-linux_armv7l.whl from https://markwal.github.io/wheelhouse/cryptography-2.5-cp27-none-linux_armv7l.whl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (2.5)
> 2020-08-11 10:17:17,309 > Requirement already satisfied: cffi@ https://markwal.github.io/wheelhouse/cffi-1.12.1-cp27-none-linux_armv7l.whl from https://markwal.github.io/wheelhouse/cffi-1.12.1-cp27-none-linux_armv7l.whl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (1.12.1)
> 2020-08-11 10:17:17,309 > Requirement already satisfied: Pillow@ https://markwal.github.io/wheelhouse/Pillow-5.4.1-cp27-none-linux_armv7l.whl from https://markwal.github.io/wheelhouse/Pillow-5.4.1-cp27-none-linux_armv7l.whl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (5.4.1)
> 2020-08-11 10:17:17,310 > Requirement already satisfied: SocketIO-client in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (0.7.2)
> 2020-08-11 10:17:17,311 > Requirement already satisfied: pyopenssl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (19.1.0)
> 2020-08-11 10:17:17,312 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.0.0)
> 2020-08-11 10:17:17,312 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (5.7.2)
> 2020-08-11 10:17:17,313 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.5.4)
> 2020-08-11 10:17:17,314 > Requirement already satisfied: monotonic<2,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.5)
> 2020-08-11 10:17:17,315 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.0.7)
> 2020-08-11 10:17:17,315 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (5.3.1)
> 2020-08-11 10:17:17,316 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.10.9)
> 2020-08-11 10:17:17,317 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.18.2)
> 2020-08-11 10:17:17,318 > Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (5.2.1)
> 2020-08-11 10:17:17,318 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.7.19)
> 2020-08-11 10:17:17,319 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (2.24.0)
> 2020-08-11 10:17:17,319 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (7.1.2)
> 2020-08-11 10:17:17,320 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.4)
> 2020-08-11 10:17:17,321 > Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.10.3)
> 2020-08-11 10:17:17,321 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.1.0)
> 2020-08-11 10:17:17,322 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.5.0.1)
> 2020-08-11 10:17:17,323 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (2.11.2)
> 2020-08-11 10:17:17,324 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.2)
> 2020-08-11 10:17:17,325 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.5.0)
> 2020-08-11 10:17:17,325 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.1.2)
[... cut ...]
> 2020-08-11 10:17:19,845 > Successfully built OctoPrint-PolarCloud
> 2020-08-11 10:17:21,022 > Installing collected packages: OctoPrint-PolarCloud
> 2020-08-11 10:17:21,023 > Found existing installation: OctoPrint-PolarCloud 1.9
> 2020-08-11 10:17:21,024 > Uninstalling OctoPrint-PolarCloud-1.9:
> 2020-08-11 10:17:21,025 > Successfully uninstalled OctoPrint-PolarCloud-1.9
> 2020-08-11 10:17:21,025 > Successfully installed OctoPrint-PolarCloud-1.9
> 2020-08-11 10:20:48,224 > Cleaning up plugin polarcloud...

wjsteele, 2020-08-11:
Hmmm... it's pulling the correct ones... are you 100% sure you followed Marks instructions.  He has a comment down below that I think might have been missed. 

See this comment: https://github.com/markwal/OctoPrint-PolarCloud/issues/38#issuecomment-397842628

reuter, 2020-08-11:
OK, I did miss that. Tried that, but it's still not working.
I followed the steps in 43 as well, still not working :(

Here's the logs for the latest attempt.
> 2020-08-11 18:56:09,830 * Installation needs to process external dependencies, that might make it take a bit longer than usual depending on the pip version
> 2020-08-11 18:56:09,847   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpUyBz1a/OctoPrint-PolarCloud-master.zip --no-cache-dir --process-dependency-links
> 2020-08-11 18:56:14,175 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> 2020-08-11 18:56:14,177 > Processing /tmp/tmpUyBz1a/OctoPrint-PolarCloud-master.zip
> 2020-08-11 18:56:16,603 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (1.4.2)
> 2020-08-11 18:56:17,114 ! DEPRECATION: Dependency Links processing has been deprecated and will be removed in a future release. A possible replacement is PEP 508 URL dependencies. You can find discussion regarding this at https://github.com/pypa/pip/issues/4187.
> 2020-08-11 18:56:20,658 > Requirement already satisfied: cryptography@ https://markwal.github.io/wheelhouse/cryptography-2.5-cp27-none-linux_armv7l.whl from https://markwal.github.io/wheelhouse/cryptography-2.5-cp27-none-linux_armv7l.whl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (2.2.2)
> 2020-08-11 18:56:20,659 > Requirement already satisfied: cffi@ https://markwal.github.io/wheelhouse/cffi-1.12.1-cp27-none-linux_armv7l.whl from https://markwal.github.io/wheelhouse/cffi-1.12.1-cp27-none-linux_armv7l.whl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (1.11.5)
> 2020-08-11 18:56:20,660 > Requirement already satisfied: Pillow@ https://markwal.github.io/wheelhouse/Pillow-5.4.1-cp27-none-linux_armv7l.whl from https://markwal.github.io/wheelhouse/Pillow-5.4.1-cp27-none-linux_armv7l.whl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (6.2.2)
> 2020-08-11 18:56:20,661 > Requirement already satisfied: SocketIO-client in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (0.7.2)
> 2020-08-11 18:56:20,662 > Requirement already satisfied: pyopenssl in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-PolarCloud==1.9) (18.0.0)
> 2020-08-11 18:56:20,662 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.0.0)
> 2020-08-11 18:56:20,663 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (5.7.2)
> 2020-08-11 18:56:20,664 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.5.4)
> 2020-08-11 18:56:20,664 > Requirement already satisfied: monotonic<2,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.5)
> 2020-08-11 18:56:20,665 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.0.7)
> 2020-08-11 18:56:20,666 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (5.3.1)
> 2020-08-11 18:56:20,666 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.10.9)
> 2020-08-11 18:56:20,667 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.18.2)
> 2020-08-11 18:56:20,668 > Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (5.2.1)
> 2020-08-11 18:56:20,669 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.7.19)
> 2020-08-11 18:56:20,669 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (2.24.0)
> 2020-08-11 18:56:20,670 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (7.1.2)
> 2020-08-11 18:56:20,671 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.4)
> 2020-08-11 18:56:20,672 > Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (0.10.3)
> 2020-08-11 18:56:20,672 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.1.0)
> 2020-08-11 18:56:20,673 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (1.5.0.1)
> 2020-08-11 18:56:20,674 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.9) (2.11.2)
[... cut ...]
> 2020-08-11 18:56:21,717 > Installing collected packages: OctoPrint-PolarCloud
> 2020-08-11 18:56:21,718 > Found existing installation: OctoPrint-PolarCloud 1.9
> 2020-08-11 18:56:21,718 > Uninstalling OctoPrint-PolarCloud-1.9:
> 2020-08-11 18:56:21,719 > Successfully uninstalled OctoPrint-PolarCloud-1.9
> 2020-08-11 18:56:21,720 > Running setup.py install for OctoPrint-PolarCloud: started
> 2020-08-11 18:56:24,162 > Running setup.py install for OctoPrint-PolarCloud: finished with status 'done'
> 2020-08-11 18:56:24,163 > Successfully installed OctoPrint-PolarCloud-1.9

Onehunterguy, 2020-08-24:
Reinstalled and the plug in now works...thanks Guys!

closed 2020-09-17 by markwal