markwal/OctoPrint-PolarCloud #51: Polar cloud Plugin installs successfully but does not appear in plugins

wsager42, 2020-08-12:
Hello 
Just like issue #38 I'm having the same problem.
I entered all the commands in my raspberry pi after connecting to it through ssh

`source ~/oprint/bin/activate
pip uninstall cryptography
pip uninstall cffi
pip uninstall Pillow
pip install cffi=1.11.5
pip install cryptography==2.2.2
pip install Pillow`

Even after using the commands Polar Cloud successfully installs but does not appear in plug in manager after octroprint restarts
my printer is the Ender 3 V2, and I'm using a raspbery pi 4
Octoprint 1.4.2
Python 2.7.16
Octopi 0.17.0

I saw somewhere on Github that Cura Engine Legacy needed to be installed but I wasn't sure how to configure it.

Your help will be greatly appreciated I've been trying to get it to work all week long!

userix, 2020-08-15:
I am having the same issue too.  Plugin says installed successfully, but upon reboot, the plugin isn't listed and when I go to install it again, it shows the INSTALL button as if I haven't installed it at all.

evan2645, 2020-08-16:
I'm seeing the same thing. I applied all available updates to the pi via apt-get upgrade, rebooted, still no luck.

My octopi install is brand new, just a couple weeks old. I have no other plugins installed, this is my first.

markwal, 2020-08-26:
Fixed in 1.10

closed 2020-08-26 by markwal