markwal/OctoPrint-PolarCloud #54: Delta printer can't only get registered as cartesian

gdachs, 2020-10-08:
The dialog for the plugin in OctoPrint has only the setting cartesian. 
I can later change the bed in PolarCloud, but this settings don't persists.

vmorrisPolar, 2020-12-17:
Hi @gdachs, sorry to hear. Upon initial setup, the Plugin should indeed have Delta as an option, it's our Printer Type: 21, from there you'll be able to choose from 4 default config files. If you let me know what email you log into the cloud with, I should be able to fix it for you.

gdachs, 2020-12-17:
I just have sent you an email.

vmorrisPolar, 2020-12-17:
Printer record was fixed. To reduce confusion, we may make improvements to the plugin for setup, only allowing the user to choose Cartesian(rectangular)/Delta(circular) from the printer type list, and then choose from an updated version of the Manufacturers list.

closed 2020-12-17 by vmorrisPolar

gdachs, 2020-12-17:
You are too fast with closing this. Checked the printer in the cloud, it had now the right bed shape, only the dimensions were off, but I have not touched them.
Then I changed to my TronXY printer and back to the delta and your changes were lost.
On the other side, this is not an issue with the plugin.

gdachs, 2020-12-17:
All this settings are already known by Octoprint, why to maintain a manufacturers list if you just can ask Octoprint?

vmorrisPolar, 2020-12-17:
The dimensions are handled in the frontend. I'm trying to limit issues to only those concerned with advancement of the lugin itself - I'm happy to work with you on the cloud side of things in a forum appropriate for it.

To my knowledge, Octoprint asks the user to enter their printer type in text fields called "Make" and "Model", we'd be looking to collect the info in more reliable formats to avoid identifying "Prusa Mk3" and "Prusa Mk.3" as 2 different printers.