markwal/OctoPrint-PolarCloud #55: Failed to register OctoPrint with the Polar Cloud. The plugin failed to generate a signing key. Please see troubleshooting tips in the README.

FltLtCrunchy, 2020-11-01:
i have just tried to set up polar cloud with octoprint windows. and i keep gettin this message "Failed to register OctoPrint with the Polar Cloud. The plugin failed to generate a signing key. Please see troubleshooting tips in the README." i have tried changing my pin and reinstalling the plugin but nothing.
please help

Aeroheadcww, 2020-11-10:
Same: Getting this error trying to register a printer with the polar cloud plugin: "Failed to register OctoPrint with the Polar Cloud. The plugin failed to generate a signing key"  I have gone through the troubleshooting and read through previous posts.  I have updated all of my python files.  No results.  Pls help.

markwal, 2020-11-19:
Hmmm...  What does octoprint.log say?

markwal, 2020-11-19:
For @FltLtCrunchy , I'm sure it is due to running on Windows.  More than likely it is because the cryptography module isn't installed and/or isn't generating the proper key.  In addition, Windows doesn't have ssh-keygen installed by default.  But it would be good to see the octoprint.log to see where it is going wrong for you.

For @Aeroheadcww , I'm not sure if you're in the same situation.  Are you installing on Windows also?

wyannick, 2020-12-20:
I have exactly the same problem as the topic starter. I have an account on polar cloud and I know the Pin. When I try to add my printer in octoprint it says: The plugin failed to generate a signing key. Please see troubleshooting tips in the README. I am using a windows computer.

Here are some logs:
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3960000512
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.2.4
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Marlin TH3D U1.R2.B5

2020-12-20 20:31:36,079 - octoprint.access.users - INFO - Logged in user: Yannick
2020-12-20 20:31:37,720 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.178.18
2020-12-20 20:31:37,748 - octoprint.server.util.flask - INFO - Passively logging in user Yannick from ::ffff:192.168.178.18
2020-12-20 20:31:37,748 - octoprint.access.users - INFO - Logged in user: Yannick
2020-12-20 20:31:39,231 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
2020-12-20 20:31:40,124 - octoprint.server.util.sockjs - INFO - User Yannick logged in on the socket from client ::ffff:192.168.178.18
2020-12-20 20:31:44,633 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.178.18
2020-12-20 20:32:16,166 - octoprint.plugins.polarcloud - ERROR - Unable to generate or access key.
2020-12-20 20:32:16,166 - octoprint.plugins.polarcloud - INFO - Can't register because unable to generate signing key

wyannick, 2020-12-20:
I have exactly the same problem as the topic starter. I have an account on polar cloud and I know the Pin. When I try to add my printer in octoprint it says: The plugin failed to generate a signing key. Please see troubleshooting tips in the README. I am using a windows computer.

Here are some logs:
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3960000512
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.2.4
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Marlin TH3D U1.R2.B5

2020-12-20 20:31:36,079 - octoprint.access.users - INFO - Logged in user: yawo
2020-12-20 20:31:37,720 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.179.18
2020-12-20 20:31:37,748 - octoprint.server.util.flask - INFO - Passively logging in user yawo from ::ffff:192.168.179.18
2020-12-20 20:31:37,748 - octoprint.access.users - INFO - Logged in user: yawo
2020-12-20 20:31:39,231 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
2020-12-20 20:31:40,124 - octoprint.server.util.sockjs - INFO - User yawo logged in on the socket from client ::ffff:192.168.179.18
2020-12-20 20:31:44,633 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.179.18
2020-12-20 20:32:16,166 - octoprint.plugins.polarcloud - ERROR - Unable to generate or access key.
2020-12-20 20:32:16,166 - octoprint.plugins.polarcloud - INFO - Can't register because unable to generate signing key

AdrianBosze, 2021-01-05:
I'm having the same issue. Octopi is 0.18.0 64bit 
Just this. No other info
2021-01-05 21:49:13,809 - octoprint.plugins.polarcloud - INFO - Can't register because unable to generate signing key
2021-01-05 21:49:14,524 - octoprint.plugins.polarcloud - ERROR - Unable to generate or access key.
2021-01-05 21:49:14,526 - octoprint.plugins.polarcloud - INFO - Can't register because unable to generate signing key
2021-01-05 21:49:14,711 - octoprint.plugins.polarcloud - ERROR - Unable to generate or access key.
2021-01-05 21:49:14,712 - octoprint.plugins.polarcloud - INFO - Can't register because unable to generate signing key

Aeroheadcww, 2021-01-05:
Octopi/Octoprint is installed on a Rasperry pi 3.  Accessing from a Chrome browser on Windows to try and install the plugin.  I don't have log files because I've given up and erased everything.

kenfager, 2021-01-26:
Same issue here. RPI 4 running latest Raspbian and Octoprint. Not sure where to look or how to proceed. 

OctoPrint Log
2021-01-26 14:00:15,494 - octoprint.plugins.polarcloud - ERROR - Unable to generate or access key.
2021-01-26 14:00:15,495 - octoprint.plugins.polarcloud - INFO - Can't register because unable to generate signing key

Plugin Log
2021-01-26 13:59:04,544   /home/pi/OctoPrint/venv/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp19bchbw6/OctoPrint-PolarCloud-master.zip --no-cache-dir
2021-01-26 13:59:08,155 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2021-01-26 13:59:08,156 > Processing /tmp/tmp19bchbw6/OctoPrint-PolarCloud-master.zip
2021-01-26 13:59:08,156 > Requirement already satisfied: OctoPrint in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint-PolarCloud==1.12) (1.5.3)
2021-01-26 13:59:08,157 > Requirement already satisfied: Pillow in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint-PolarCloud==1.12) (8.1.0)
2021-01-26 13:59:08,157 > Requirement already satisfied: SocketIO-client in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint-PolarCloud==1.12) (0.7.2)
2021-01-26 13:59:08,157 > Requirement already satisfied: pyopenssl in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint-PolarCloud==1.12) (20.0.1)
2021-01-26 13:59:08,157 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (7.1.2)
2021-01-26 13:59:08,157 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (2.0)
2021-01-26 13:59:08,158 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (1.0.0)
2021-01-26 13:59:08,158 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (0.5.0)
2021-01-26 13:59:08,158 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (2.11.2)
2021-01-26 13:59:08,158 > Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (2020.8.7)
2021-01-26 13:59:08,158 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (2020.9.23)
2021-01-26 13:59:08,158 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (5.3.1)
2021-01-26 13:59:08,158 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (1.4)
2021-01-26 13:59:08,159 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (0.1.1)
2021-01-26 13:59:08,159 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (0.6.0)
2021-01-26 13:59:08,159 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (1.0.7)
2021-01-26 13:59:08,159 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (1.1.2)
2021-01-26 13:59:08,159 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (1.2)
2021-01-26 13:59:08,159 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (0.18.2)
2021-01-26 13:59:08,159 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (1.1.0)
2021-01-26 13:59:08,160 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (3.1.1)
2021-01-26 13:59:08,160 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (1.1.1)
2021-01-26 13:59:08,160 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (0.8.0)
2021-01-26 13:59:08,160 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (0.10.9)
2021-01-26 13:59:08,160 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (1.6.1)
2021-01-26 13:59:08,160 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.12) (5.8.0)
[... cut ...]
2021-01-26 13:59:08,164 > Requirement already satisfied: pycparser in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from cffi>=1.12->cryptography>=3.2->pyopenssl->OctoPrint-PolarCloud==1.12) (2.20)
2021-01-26 13:59:08,165 > Using legacy 'setup.py install' for OctoPrint-PolarCloud, since package 'wheel' is not installed.
2021-01-26 13:59:08,165 > Installing collected packages: OctoPrint-PolarCloud
2021-01-26 13:59:08,165 > Running setup.py install for OctoPrint-PolarCloud: started
2021-01-26 13:59:09,167 > Running setup.py install for OctoPrint-PolarCloud: finished with status 'done'
2021-01-26 13:59:09,168 > Successfully installed OctoPrint-PolarCloud-1.12

markwal, 2021-01-26:
Hmm I wonder why I can't reproduce this. Perhaps it is because I always set up ssh with ssh-agent as a matter of course. I'll try skipping that and see where I get.

markwal, 2021-01-28:
OK.  I've got this figured out.  And yes, I made a stupid mistake (over and over again) trying to repro this earlier.  It's really obvious now that I've got it figured out.  If the plugin has never generated a key on a fresh python3 system (0.18), it can't write the key out because it's an array of bytes and the file isn't opened in binary mode.  Python 2 doesn't care because bytes and str are compatible.

I have a fix that appears to work in my testing, but I need to verify (properly) that I didn't break the plugin for existing python2 folks.

closed 2021-01-28 by markwal

markwal, 2021-01-28:
Thanks for the reports.  Sorry it took me so long to figure out.

markwal, 2021-01-28:
Also, I added better logging so a similar issue will be more obvious more quickly.

kenfager, 2021-01-28:
Hey! I appreciate the detailed follow up. I think you may have this one solved because I did have python3 installed on a new system. 

Looking forward to giving the update a try.

wyannick, 2021-01-31:
It finally works! Thanks!

commit 58d96a4c (2021-01-28): Fix exception creating key file under Python3

This should fix #55
Also, remove stale dependencies for Python 2.  Hey all, if you haven't
already, you may want to backup your octoprint settings (recent versions
have a feature for that) and make a new OctoPi 0.18 image for your pi.
It'll move you to Python 3 and give you a fresh OS too.