markwal/OctoPrint-PolarCloud #56: No Stream on Polar3D.com

J-Cloos, 2020-11-28:
Hey all,

I'm at my wit's end TBH. Have gone through the entire installation process. Have tried the suggested solutions on this post: https://github.com/markwal/OctoPrint-PolarCloud/issues/4 (resetting my router and restarting OctoPrint - which was kind of annoying, as I had to find out the new IP. 

I go into OctoPrint Settings > Features> > Webcam & Timelapse menu, add my IP to the Stream URL as follows: http://192.168.1.206/webcam/?action=stream, even tested the stream, and it worked!

But I get to the website... and there's no stream!
[image]

Is there anything to be done?

Thanks in advance!

wjsteele, 2020-11-28:
So, that's not actually an issue with Polar3D.com, but rather your internal network.  To verify that, go to any network outside your home network and navigate the the printer's page and you'll see the image.  (You can use your cell phone's data to do that... just make sure your Wifi is not connected to your house network.)

If you can see it, which I'm positive you can, then you can use the browser's Developer features to see the network traffic and what URL it's requesting and what the error it is getting is.  Generally, this is caused by your networks router blocking internal traffic.

dcnewman, 2020-11-28:
Are you using chrome for your browser?  It blocks http now when the main page is https.  There was a news article on this at polar3d.com about a month ago.  Click on the news link there.

Sent from a pay phone

wjsteele, 2020-11-28:
I found a link to the solution.

https://twitter.com/Polar3D/status/1319372758279704582

vmorrisPolar, 2020-12-04:
Hi J-Cloos, there's more info at [about.polar3d.com/cam-error](about.polar3d.com/cam-error), we're exploring work-arounds as we assume other browsers will move this direction shortly. In the meantime, I'll close the ticket. I'll report back with any updates.

closed 2020-12-04 by vmorrisPolar