markwal/OctoPrint-PolarCloud #57: Cannot install Octoprint plugin on Raspi 3

Lumics, 2021-02-10:
Installiere Plugin "Polar Cloud" von https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpxqf96l4l/OctoPrint-PolarCloud-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpxqf96l4l/OctoPrint-PolarCloud-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-PolarCloud==1.13) (1.5.3)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.6.0)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (3.1.1)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.0.0)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.5.0)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.18.2)
Requirement already satisfied: feedparser<7,>=6.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (6.0.2)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (3.5)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.0.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (2.0)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.57.0)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.6.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (2020.11.13)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.2)
Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (2020.8.7)
Requirement already satisfied: zeroconf<0.25,>=0.24 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.24.5)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.0.7)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.10.4)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (5.3.1)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.4)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.1.1)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (2.8.5)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (7.1.2)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.1.2)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (2.25.1)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (5.1.1)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.4.21)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (4.0)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.1.0)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (2021.2.4)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.2.0)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (2.11.2)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.8.0)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.10.9)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.12.1)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.1.5.post0)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (5.8.0)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (0.19.5)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.13) (1.1.1)
Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python3.7/site-packages (from rsa==4.0->OctoPrint->OctoPrint-PolarCloud==1.13) (0.4.8)
Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog==0.10.4->OctoPrint->OctoPrint-PolarCloud==1.13) (0.1.2)
Requirement already satisfied: sgmllib3k in /home/pi/oprint/lib/python3.7/site-packages (from feedparser<7,>=6.0.2->OctoPrint->OctoPrint-PolarCloud==1.13) (1.0.0)
Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->OctoPrint-PolarCloud==1.13) (2.0)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->OctoPrint-PolarCloud==1.13) (2.9.0)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->OctoPrint-PolarCloud==1.13) (2020.5)
Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python3.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->OctoPrint-PolarCloud==1.13) (51.1.2)
Requirement already satisfied: idna<3,>=2.5 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoPrint-PolarCloud==1.13) (2.10)
Requirement already satisfied: certifi>=2017.4.17 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoPrint-PolarCloud==1.13) (2020.12.5)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoPrint-PolarCloud==1.13) (1.26.2)
Requirement already satisfied: chardet<5,>=3.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoPrint-PolarCloud==1.13) (4.0.0)
Requirement already satisfied: six in /home/pi/oprint/lib/python3.7/site-packages (from websocket-client<1,>=0.57->OctoPrint->OctoPrint-PolarCloud==1.13) (1.15.0)
Requirement already satisfied: ifaddr in /home/pi/oprint/lib/python3.7/site-packages (from zeroconf<0.25,>=0.24->OctoPrint->OctoPrint-PolarCloud==1.13) (0.1.7)
Collecting Pillow
[... cut ...]
=============================DEBUG ASSISTANCE=============================
error: Can not find Rust compiler
----------------------------------------
ERROR: Failed building wheel for cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
Building wheel for cryptography (PEP 517): finished with status 'error'
Successfully built OctoPrint-PolarCloud
Failed to build cryptography
Fehler!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Lumics, 2021-02-10:
Seems to be related to this: https://github.com/pyca/cryptography/issues/5771

DigitalCardinal, 2021-02-10:
Having same issue.  Read the entire cryptography thread.  Hopefully we can get an update that will allow us to fix this.  Not sure how. but I'm happy to help anyway possible to get my OctPi back up and running with Polar.

markwal, 2021-02-23:
Holy moly.  cryptography took a dependency on rust.  Shoot, we don't have an easy way of keeping the rust compiler up to date as part of OctoPrint.

markwal, 2021-02-23:
@Lumics, @DigitalCardinal  It looks to me like there are built wheels for cryptography-3.4.6 available on piwheels.org, perhaps you were trying after cryptography had been updated, but before the wheels had been built?  Could you try again?

If that fails, would you mind trying the following, please?  From the command line:

```
source ~/oprint/bin/activate
pip install cryptography
```
If that doesn't work, could you reply here with the failure?  If it does work, try installing the plugin.  And reply back here with the results either way.  Sorry, I know this will take some time to do these steps, but I think it will really help me track down what to do about this issue.  Thank you.

Lumics, 2021-02-24:
Hi @markwal I tried reinstalling without any changes and it worked seamlessly. Could be that your first guess was right. Thanks

closed 2021-02-24 by Lumics