markwal/OctoPrint-PolarCloud #58: Video stream just shows one frame

ChopsKingsland, 2021-02-20:
Whenever Octoprint starts, it shows a few frames, but then freezes and shows a no camera symbol in the upper right... (see below) 

[image]

Any ideas on how to get it streaming?

markwal, 2021-02-23:
By "it" I presume you mean the PolarCloud UI, not the OctoPrint UI.  Does the stream work in OctoPrint?

The stream on PolarCloud is just a local reference to the OctoPrint instance, so while PolarCloud works from anywhere, the stream will only show when you are on the same LAN as OctoPrint.  Otherwise it just shows a static image that updates infrequently.

vmorrisPolar, 2021-02-23:
Hi Chops. First order of business is probably to better understand what we're seeing. Is your device on the same network as the printer? More info here: https://about.polar3d.com/camera-feed

vmorrisPolar, 2021-03-02:
@ChopsKingsland this appears to be a bug in the Polar Cloud, nothing with Octoprint or the Plugin. We'll be addressing it ASAP and I'll keep you updated, but since it's a cloud issue, probably best to close it as far as the plugin is concerned. Thanks for bringing it to our attention.

closed 2021-03-02 by vmorrisPolar