markwal/OctoPrint-PolarCloud #60: Multiple Instances Of Octoprint Assign Same Serial Number

mcurranseijo, 2021-05-27:
I have 3 instances of octoprint running on a pi with three seperate printers, The problem I found is all three are assigned the same serial number and thus show up as one printer on the polar3d interface, is there a way I can manually tell it to use a different serial number for each one or is this beyond what the plugin can/was meant to do? I tried adding one at a time and renaming it on the Polar3d side but that didn't seem to work.

vmorrisPolar, 2021-05-27:
As long as the pi's have different MAC addresses, they should all get different serial numbers. Any idea if that's not the case?

mcurranseijo, 2021-05-27:
What I meant is I'm running 3 instances on one pi, so three octoprint servers on one device, if it bases serial number off mac adress is there any reasonably simple way I can change that?

vmorrisPolar, 2021-05-27:
Unfortunately no, not at this time. Unless @alexp3d or @dnewman-polar3d have any ideas for updates to the protocol to address situations like this.

mcurranseijo, 2021-05-27:
Ok no worries I can come up with a workaround and just use the octoprint UI on the pi or something along those lines, thanks for the quick reply.