markwal/OctoPrint-PolarCloud #62: "Failed to build cryptography" - Requires rust?

vmorrisPolar, 2021-11-23:
Trying to install Polar Cloud plugin on a fresh install of Octopi 1.7.2 resulted in the following:

=============================DEBUG ASSISTANCE=============================
error: can't find Rust compiler
If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
To update pip, run:
pip install --upgrade pip
and then retry package installation.
If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
This package requires Rust >=1.41.0.
----------------------------------------
ERROR: Failed building wheel for cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

vmorrisPolar, 2021-11-29:
[image]

vmorrisPolar, 2021-11-30:
Issue didn't present itself when plugin was installed from command line using pip

markwal, 2021-11-30:
This plugin requires pyOpenSSL and uses it to both generate certificates as well as to encrypt communication to the service.  It currently requires it without version in order to consume the latest that presumably has the most security flaws fixed.  The risk, of course, is that the pyOpenSSL or one of its dependencies can break us.

**What broke?**
In this case, one of its dependencies, cryptography, has taken a dependency on a compiler, rustc, that isn't included in the pi image.  This happened some time ago (a year or so?), but pops up and breaks install from time to time.

**Why doesn't it break us all the time?**
Currently, someone is maintaining a binary wheel that includes those needed for the raspberry pi.  As long as a wheel is available for the platform you are running on, pip uses that instead of building it itself.  So... if the binary wheel maintainer has built and uploaded the current version of cryptography, then installing the plugin will download that and will skip building cryptography itself.  If not, then when it tries to build it, it will discover that there isn't a rust compiler installed.

So if you wait a day or so after failing to install, it may just work again when that wheel becomes available.

**How do we keep it from breaking for a day or two everytime cryptography updates?**
Ah, this is the trick.  Some possibilities:
* We could pin the version of pyOpenSSL to last known working, but then we'll need to stay on top of every time it updates and update the plugin also.
* We could write some code in setup.py that tries to discover the latest available cryptography wheel and then figure out what pyOpenSSL to ask for to pick that one up.  Or maybe something dumber.
* Or maybe there is some option that tells setup that you only want to consider wheels.

vmorrisPolar, 2021-12-15:
Hey Mark. I talked a little with Bill Steele (@wjsteele) about this. His 0.02 was that it might make sense to implement multiple solutions.. so at least 2 out of 3 of the possibilities you mention.

If you are too busy to do the work, we have a few members that can chip in, just need to figure out what makes the most sense (now and moving forward). Would appreciate your counsel. Whatever you recommend.

I don't have a full grasp on what all this issue can affect - i.e. octopi installs only via plugin manager, builds from scratch on top of raspbian, or more.

markwal, 2021-12-15:
It should affect all installs of the plugin, from the command line, plugin manager, etc.  The difference between success and failure is really just how long its been since a new release of the cryptography package.  And it may be more likely to fail on a pi since the PC wheels may be released more quickly.

markwal, 2021-12-15:
I'll see if I can code something up here in a bit.

vmorrisPolar, 2022-02-25:
@markwal is this the same issue? A different school reported this recently.

`Installing plugin "Polar Cloud" from https://github.com/markwal/OctoPrint-PolarCloud/archive/master.zip...
/usr/local/bin/python -m pip --disable-pip-version-check install [file:///tmp/tmpdb0ylogz/OctoPrint-PolarCloud-master.zip]() --no-cache-dir
Processing /tmp/tmpdb0ylogz/OctoPrint-PolarCloud-master.zip
Requirement already satisfied: OctoPrint in /usr/local/lib/python3.8/site-packages (from OctoPrint-PolarCloud==1.15) (1.7.3)
Collecting Pillow
Downloading Pillow-9.0.1.tar.gz (49.5 MB)
Collecting SocketIO-client
Downloading socketIO-client-0.7.2.tar.gz (23 kB)
Collecting pyopenssl
Downloading pyOpenSSL-22.0.0-py2.py3-none-any.whl (55 kB)
Requirement already satisfied: unidecode in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.3.2)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (2.8.5)
Requirement already satisfied: emoji<2,>=1.4.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.6.3)
Requirement already satisfied: markdown<3.2,>=3.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (3.1.1)
Requirement already satisfied: pkginfo<2,>=1.7.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.8.2)
Requirement already satisfied: regex in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (2022.1.18)
Requirement already satisfied: psutil<6,>=5.8 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (5.9.0)
Requirement already satisfied: requests<3,>=2.26.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (2.27.1)
Requirement already satisfied: Jinja2<3,>=2.11.3 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (2.11.3)
Requirement already satisfied: tornado<7,>=6 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (6.1)
Requirement already satisfied: future<1,>=0.18.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (0.18.2)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (3.5)
Requirement already satisfied: websocket-client<1,>=0.59 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (0.59.0)
Requirement already satisfied: netifaces<1,>=0.11 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (0.11.0)
Requirement already satisfied: flask<2,>=1.1.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.1.4)
Requirement already satisfied: wrapt<1.13,>=1.12.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.12.1)
Requirement already satisfied: cachelib<0.2,>=0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (0.1.1)
Requirement already satisfied: pylru<2,>=1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.2.0)
Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (0.8.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.0.1)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (2021.10.11)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (2021.2.23)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.0.0)
Requirement already satisfied: zipstream-new<1.2,>=1.1.8 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.1.8)
Requirement already satisfied: zeroconf<0.34,>=0.33 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (0.33.4)
Requirement already satisfied: colorlog<6,>=5.0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (5.0.1)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (1.1.0)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (0.5.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (2.0)
Requirement already satisfied: Click<8,>=7.1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (7.1.2)
Requirement already satisfied: sarge==0.1.6 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (0.1.6)
Requirement already satisfied: immutabledict<3,>=2.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-PolarCloud==1.15) (2.2.1)
[... cut ...]
command: /usr/local/bin/python /tmp/pip-standalone-pip-cu6g4hby/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-xqte31cj/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.12; platform_python_implementation != '"'"'PyPy'"'"'' 'setuptools-rust>=0.11.4'
cwd: None
Complete output (108 lines):
Collecting setuptools>=40.6.0
Using cached setuptools-60.9.3-py3-none-any.whl (1.1 MB)
Collecting wheel
Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Collecting cffi&gt`

markwal, 2022-02-25:
Yes.  Sorta.  Though for one of the cryptography'dependencies instead. cffi

One way we could workaround these issues would be to provide our own Octopi image like the fellow you put me into contact with does.

Main drawbacks being youd need somebody to monitor the thing (it ops) it'd probably cost a few monthly bucks in egress to host a file that large.

Sent from Surface Duo

vmorrisPolar, 2022-02-28:
I don't have a problem hosting or doing that. I would just need to know the best way to do it and what all packages and dependencies we'd need.